volumio/Volumio2 #578: Web UI not reachable (failed to connect)

DaveGahan, 2016-07-23:
Hello, I just switched my PI B+ from Volumio 1.55 to latest RC2 hotfix image by flashing it on the memory card.

I can't access web UI from any computer (linux, windows, wired or not). SSH works (with putty or JuiceSSH on my android phone, not with regular ubuntu terminal ssh command, it gets stuck on login).

www folder seems to be ok on the PI and, thinking it could have something to do with my rooter (ISP "box") I rebooted it but nothing changes.

Do you have any idea to help me please ?

PS: Volumio 1.55 or RuneAaudio UIs work flawlessly

mlorenzon, 2016-07-24:
Hi, I'm having the same issue with RC2 on both Pi B+ and 3. SSH is fine but both volumio.local and IP address time out.

xipmix, 2016-07-24:
Some questions:
  * does the startup sound play?
  * is the Pi connected over wireless or ethernet?

It would be helpful to pastebin the file /var/log/volumio.log. First do a reboot and wait say 1 minute or until the volumio 'startup sound' plays. You should be able to get it with scp:

    otherbox$ scp volumio@volumio:/var/log/volumio.log

winscp should work as well.

DaveGahan, 2016-07-24:
I'm truly sorry, it seems like I wasn't patient enough, it works a few minutes after Pi start (I'm used to 1.55 speed). I got an Hifiberry digi+ connected to an external DAC (toslink). If you need any feedback don't hesitate to ask.

Thanks for this great job :-)

mlorenzon, 2016-07-24:
Thanks for your response. I had flashed the SD cards with dd. I reflashed them with Etcher and was able to access the UI.

closed 2016-07-25 by apinner

wsloand, 2016-07-31:
I'm having the same trouble.  I did a fresh flash of the Volumio2 image (volumio-0.976-2016-07-28-pi.img), and after the flash, I had trouble where I could not load the main web page.  I get a page titled "{{pageTitle}}" with little else present, Chrome indicates that it's still loading the page, and nothing else happens.

I have a bit more detail that when I first boot, I can access ssh, but about 1-2 minutes later, the current ssh connection drops, and I cannot reconnect.

I have tried flashing both with Win32DiskImager and with Etcher.  Both give similar results.

wsloand, 2016-07-31:
As I'm looking at it a bit more, I see that just before I lose my connection via ssh, I can see via top that the node process is taking ~72% of the available memory.  Doing a bit more investigation, I have a RPi 2 B+ (revision 1.0) with 512 MB RAM with a HifiBerry AMP+ connected to it.

It appears that trying to access the web page is what kills the ssh session.  I can still ping the RPi.