volumio/Volumio2 #580: Wifi doesn't work on Rpi using RC2

heis2201, 2016-07-24:
/etc/network/interfaces shows a line
```
allow-hotplug wlan0
iface wlan0 inet manual
```
while in volumio's UI it is set for auto-address. Also, in UI, can't connect to wifi. `ifconfig` doesn't show that network address is received...

heis2201, 2016-07-25:
Had to change *interfaces* manually to 

```
allow-hotplug wlan0
iface wlan0 inet dhcp

```
in order to establish a wifi connection.

kaffeeringe, 2016-08-15:
I had the same problem. And the solution also works for me.

kaffeeringe, 2016-08-15:
The UI still shows a wired connection with 10Mb/s and IP 192.168.8.8 - altought the cable is disconnected and I already rebooted the system.

heis2201, 2016-08-15:
Same here but I don't bother too much... as long as wifi connection is reliable and stable.

volumio, 2016-08-15:
As per wi-fi issue, did you had this trouble while trying to connect from UI?

heis2201, 2016-08-16:
@volumio : Sorry, never answered your above question. Yes, when connected via ethernet, I could access my Volumio but I was not able to establish a wifi connection. Tried it several times. Then I took a look at `interfaces` and `wlan0` would always remain in `manual`. Once I changes that value to `dhcp`, my wifi connection is working just fine.

closed 2017-06-02 by volumio