volumio/Volumio2 #581: adding Web Radio needs explanation

heis2201, 2016-07-24:
When adding a Web Radio stream, how do I add local m3u/pls file?

apinner, 2016-07-25:
IIRC you don't any more. Radio stations are added to a json file.

heis2201, 2016-07-25:
I see, but I don't like to add all my custom stations manually (by creating a new station and point to the remote .m3u or .pls). How about an import function of the old radio files?

And what if you push an update to the .json radio library? Will it kill my custom stations? Where is the .json file stored? 
Thanks

apinner, 2016-07-26:
There will be a backup function added at some point that will allow you to download a complete configuration backup from volumio that can be imported into a new instance.

The file is located at /data/favourites/my-web-radio

Yes you can update and edit it using nano.

heis2201, 2016-07-26:
I see how they are all in the .json file but I would like to suggest an import feature of .m3u or .pls files, because that's the format that one would use to import radio stations.

Not sure if this could be implemented as plugin.

Moreover, I'd like to suggest to import the vast list of internet-radios found here: https://github.com/jprjr/internet-radio-streams

volumio, 2016-07-26:
Yes we'll add the import function like we did with previous volumio

heis2201, 2016-07-26:
Great! Thanks!

buzink, 2016-12-26:
Would it not be better to just have a +-button in the Browse-tab which you can click and than paste an url and type a name?

heis2201, 2017-02-21:
Dear friends,
is there any update in this regards? I am still looking for a function to "mass import" .m3u and .pls into Volumio2. (For example, how to use the good selection of webradios presented in Volumio 1.5x, in addition to my personal 20+ favorites?)

xipmix, 2017-04-06:
I think  this is the same issue as #334

smoneck, 2017-09-30:
I had the same problem today. I have a long `.m3u`-file with stream I use on different computers that is being updated from time to time.

As a workaround, I wrote a simple script [m3u2volumio](https://github.com/smoneck/m3u2volumio) to convert standard and extended `.m3u` files into volumio-readable syntax. It can be directly used in combination with SSH and just relies on `python3 >= 3.1`:

    m3u2volumio playlist.m3u | ssh volumio@volumiocomputer 'cat > /data/playlist/newplaylistname'

Let me know if you have any trouble with it. Maybe I add a `volumio2m3u` version just in case ...