volumio/Volumio2 #582: Appearance settings are not clear/intuitive

heis2201, 2016-07-24:
Appearance settings are not clear/intuitive, at least on my UI (Firefox 47). 
See screenshot [here](http://imgur.com/a/KjDOw).
Additionally, I think, they deserve a "reset" button that loads factory defaults for design.

volumio, 2016-07-24:
When you saw the page probably, the thumbnails were not generated yet.
Which pi models do you have? 
Could you try to reload the page and see what happens?

apinner, 2016-07-25:
Your page clearly hasn't loaded properly.

There are only two settings that you can change, I think a reset button is a bit excessive. Especially when the default background is called default.

heis2201, 2016-07-25:
@volumio : I am on RPI B+. Already tried to reload the page. Nothing changed. 

And yes, I agree with @apinner : When this page is completely loaded, there will be no need for a reset button.

closed 2016-07-26 by apinner

heis2201, 2016-07-26:
So, what's the solution to fully load the page?

volumio, 2016-07-26:
It should load automatically everytime, I'm investigating the issue which seems related to RPI B+ maybe being too slow to achieve this in a timely manner