volumio/Volumio2 #584: Rename Play into "Add & play"

heis2201, 2016-07-24:
Clicking *Play* adds the song to Queue and plays it. Why not naming the function that way? 

In Volumio 1.55, it was called *Add & Play*, and that's exactly what it does.

apinner, 2016-07-25:
This seems to be in complete contradiction to https://github.com/volumio/Volumio2/issues/560

Will have to investigate.

volumio, 2016-07-25:
I would say that we'll keep the name play, but we'll also add replace and play

plesatejvlk, 2016-07-25:
Perfect! I personally think the "replace & play" function is ingenious and should be a standard part of any music player with queue. In another words: yes, please add the "replace & play" to the list of options.:)

heis2201, 2016-07-25:
I agree. It would be perfect to have a *Play* button (that just appends to existing Queue) and *Replace & Play*. Thanks a lot for considering this!

volumio, 2016-07-26:
Done! 
https://github.com/volumio/Volumio2/commit/4f39b7b5bf71368de467e2b5d18dfcde14162ba7

closed 2016-07-28 by volumio