volumio/Volumio2 #588: Support for USB Wifi Adapter MT7601U (Ralink Technology, Corp.)

heis2201, 2016-07-26:
Hi,
I bought a USB Wifi adapter with external antenna (Ralink Technology, Corp. **MT7601U** Wireless Adapter) to make volumio's wifi connectiion more stable.

Surprisingly, this chipset is not supported out-of-the-box. Has anyone any idea how to add the corresponding kernel module (works perfectly fine on Fedora with kernel 4.5 and 4.6).

(While troubleshooting, why is `lsusb` and `lspci` not available on Volumio's debian/raspbian?

volumio, 2016-07-26:
try with sudo , probably we miss firmwares. Do you know if they are available on raspbian?

balbuze, 2016-07-26:
Yes I think we really miss those commands. Someone on the forum have a similar problem and without lspci or lsusb it'not easy to help

volumio, 2016-07-26:
Right, fixed with this:
https://github.com/volumio/Build/commit/b59d0c24847952fc2c60c5e716429d365a15aeb5

heis2201, 2016-07-26:
Don't know if firmware available on raspbian but I can research it later today.

pgpGithub, 2017-08-27:
My MT7601U works out of box with raspian but not with volumio 2.246

macmpi, 2017-08-28:
There's a pending PR fixing it: https://github.com/volumio/Build/pull/221

macmpi, 2017-08-29:
I guess this one can be closed now

heis2201, 2017-08-30:
Yes, it could be closed, PR has been merged: https://github.com/volumio/Build/pull/221

[Duplicated issue](https://github.com/volumio/Volumio2/issues/1080) has already been closed. Thank you very much for fixing this issue!

closed 2017-08-30 by heis2201