volumio/Volumio2 #589: catalan translation

nosferatujr, 2016-07-26:
Hi Guys, i produced this translation. Please consider to add. Thanks
https://gist.github.com/nosferatujr/00e5565f9451e417f6afc9d42f831f62

apinner, 2016-07-26:
Thank you @nosferatujr 

I have added those translations for you. Just as a note, if you edit in future we use two spaces as a tab. If you use one of the common text editors then there is a editorconfig here https://github.com/volumio/Volumio2-UI/blob/master/.editorconfig that should configure that for you. For the most part it was fine, just a few random tabs.

closed 2016-07-26 by apinner

volumio, 2016-07-26:
Molt bè!!! Moltes gràcies 

fins aviat 

mi

nosferatujr, 2016-07-26:
Un plaer, 
apinner  , Sorry for the mistake, I will take in consideration the tabs on the future