volumio/Volumio2 #591: Playlist controls don't seem to work on handheld devices

Saiyato, 2016-07-26:
I'm talking about these:
[image]

Which can be quite difficult when you want to play a specific playlist (e.g. Spotify). Because there is no more play/add/replace option in the menu for either albums (mpd)/playlist (all)/webradio
[image]

Since you can't clear the list on your phone, you just add a new playlist to the mix (which is no problem when you don't shuffle, but when you want shuffle it is).

I'd look into this, but my knowledge of node.js is 0 (zero). If you need anything, logs, for me to test new versions etc. etc. let me know.

apinner, 2016-07-26:
I believe the queue buttons has been fixed already, just not released yet.

Play and Replace was discussed here https://github.com/volumio/Volumio2/issues/584 and will be added again.

We have also added the shuffle and repeat buttons to the ... menu under the playback knob so you don't have to leave the playback screen.

Saiyato, 2016-07-26:
Awesome! Is there an easy way to test this by pulling from the git? Or should I wait for releases?
Also does an icon show if for example shuffling is set to on, outside the ... menu? :)

Great work btw!

apinner, 2016-07-26:
You would probably need to build the UI or run it from gulp using the instructions on the UI repo. Probably easiest to wait for the release.

I believe it will yes.

closed 2016-07-26 by apinner