volumio/Volumio2 #593: Volumio 2 Cubieboard

jersonjunior, 2016-07-27:
Volumio 2 works with Cubietruck?

volumio, 2016-07-27:
Nope. Want to help porting to it?

jersonjunior, 2016-07-27:
Yes, how can I help?

gkkpch, 2016-07-30:
ok, study http://github.com/volumio/Build. you do not need to change the build.sh script for the time being, just have a look at the raspberry or odroid builds. each have 2 parts, the image and the config build. the image part more or less builds a bootable image like creating an .img file, mounting it and creating the necessary partitions. You can copy the first part. You need to check the next part with the cubietruck forums, you will need to copy u-boot (or use an installer in the config part) and copy the rootfs. The rootfs is built with "sudo ./build.sh -b arm", which of course needs to be run before the image script. You will find the locationin the image build script where the config part is called. the config holds all configuration actions which need a running system, this is achieved by running the script with chroot/qemu. I know, this is only a small intro, if you need more info, just pm me.

closed 2017-06-02 by volumio