volumio/Volumio2 #594: Radio stream failed to decode

heis2201, 2016-07-27:
I was trying to listen to my favorite [KCRW-Eclectic 24 ](http://www.kcrw.com/music/shows/eclectic24/#)radio but volumio said it would not be able to decode the stream.
What's the problem here?

Volumio suggests that it is playing... time is counting, but there is really no output...

Screenshots are [here](http://imgur.com/a/ULkeQ), and [here](http://imgur.com/a/Bsis9)

[image]

shoredrive, 2016-07-27:
I am also having trouble using a .m3u address to stream webradio. I get the audio tone on boot, and airplay works great, however I can not stream Radio St. Barth http://www.vtuner.com/vtunerweb/mms/m3u18177.m3u  It says playing and even shows the different song data, however no audio except for some light static that disappear when i press pause.

buzink, 2016-12-26:
Did you try using the url in the m3u-file? As a workaround, you can create a playlist file with that url in the folder "/data/playlist" following the following format:

[ { "service": "webradio", "uri": "http://2stream.dyndns.org:8000/rsb", "title": "Name of station", "albumart": "/albumart?path=%2Fplaylist%2Fradio1.jpg" } ]

You can add playlists through SSH each time or through your file explorer after adding "/data/playlist" once as a network share in "/etc/samba/smb.conf"

closed 2017-06-02 by volumio