volumio/Volumio2 #597: Volume 2 won't recognise my Synology NAS on ethernet

Bonky28, 2016-07-28:
A common problem I think- but no real easy solution -yet. I use a Mac and had no problems with v1.55. The NAS is recognised on the network on the Mac but does not appear in Vol2 (the USB drive does). I'd rather not delve too much - if at all -  into Terminal and coding. Is there an easy workaround? I've tried manually setting some of the network parameters but no luck. I am trying a factory reset to start over but that just hangs!
Thanks,
Bonky

kaffeeringe, 2016-08-18:
AFAIK Volumio2 only supports NFS shares. Have you enabled NFS as a service? Have you turned your shared folders into NFS-shares?

volumio, 2016-08-18:
No, Volumio2 supports both NFS and CIFS (samba). Please try to add nounix,sec=ntlmssp to options.

Please let me know

Bonky28, 2016-08-18:
Thanks; we’ve had no broadband for 11 days -now thankfully fixed I hope, so will test later. In the meantime I've lost the Volumio2 Sd card, so may need to buy another.

Apologies!
BW

Richard

kaffeeringe, 2016-08-18:
Can you rename CIFS in that dropdown to SMB or Samba or Windows? I thought I new at least a little bit about Samba, but I didn't know CIFS.

@Bonky28: You add these options in /etc/exports - so you have to use SSH. I don't think you can do that via the Synology UI.

closed 2017-06-02 by volumio