volumio/Volumio2 #599: After todays update of Volumio2: Can't reach neither web server nor ssh..

heis2201, 2016-07-29:
Installed update from WebUI. Feedback was positive. After restart, can't web server, can't reach ssh server.
Tried to restart already, to no avail.

Please advise what to do and how to troubleshoot.

heliosphan79, 2016-07-29:
Same here... On my rpi3 everything worked out fine. my rpi2 isn't accessible anymore and kills the node process with an out of memory notice...

nosferatujr, 2016-07-29:
Could be the same happens to me, I thing that is caused for the language (see issue #598), really I reach the volumio trough SSH, and change the language, and after the restart the web server works again.  
But after back to English, the web server appears today off again ones, but after a new reboot all works again.
Something is happening, need investigation to found the real cause….

nosferatujr, 2016-07-29:
@heliosphan79 Can you ssh your rasp2?

skikirkwood, 2016-07-29:
In installed 0.976 on a Rpi3 with no problem, but can't connect to it in any way on a Rpi B+.  It boots up and my Rpi3 sees it with the multi-room interface, but I can't select it, and with the Rpi3 turned off I can't ssh into or get the Web interface for the B+.

nosferatujr, 2016-07-29:
but not sure, Really I reboot in english few times with success, is switch language to catalan, works until reboot. on the reboot ssh works and webserver not, changing /data/configuration/miscellanea/appearance/config-json to English the e initial sound is played and webserver works again...
Anyone of the guys that experiment the problem is using different language?

heliosphan79, 2016-07-29:
@nosferatujr, no ssh access... My rpi doesn't respond to anything... Hopefully you guys can find a quick fix... I'm missing the romantic music in my bedroom ;-)

Tihmann, 2016-07-29:
same problem here. Update and now volumio is not available anymore....

Tihmann, 2016-07-30:
I am using a Rpi 2

volumio, 2016-07-30:
So you did an update? From which version?  Is there any body  willing to reflash and see if that changes things?  No quick fix since I am in holydays...

balbuze, 2016-07-30:
I did a update from 0.975 and it results a kernel panic. A fresh install works on RPI2. Haven't test other devices.

JedS, 2016-07-30:
Same here.
Update from 0.928 and got a kernel panic too.
node process seems to take too much memory as I got an 
**Out of Memory Killing process 928 (node)
Killed process 928 (node) total-vm 336164Kb**
I was able to connect to my Pi just after a reboot, and I ran top 
It confirm an Out of Memory problem, which freeze the whole system.

gik98, 2016-07-30:
Same here on a rpi b+ after fresh install.
Unable to reach web ui/ssh, although it seems to be connected and scanning ports I see 22&80&others opened.
Attaching a screen and a keyboard I notice random crashes after a couple of minutes from startup.
Here's the last line of dmesg:
`pcm512x 1-004d: No SCLK, using BCLK: -2`

As far as I know it is something concerning the I2S sample clock, but I doubt it is the cause of the failure. I noticed a huge consumption of RAM as @JedS stated, mostly related to node service.

skikirkwood, 2016-07-30:
I didn’t do an update, I flashed a card that previously had 1.55 on it with the 0.976 download that worked perfectly on the Rpi 3.  Not sure how much more I can diagnose as I couldn’t ssh in.  I did see my other Pi running Volumio 2 see it as it showed up under multi-room on its home page and the Volumio hotspot showed up.  The web interface turned the screen dark gray, but never rendered a page.

heliosphan79, 2016-07-30:
@volumio , enjoy your holidays! The bedroom-romance can wait

vostok4, 2016-07-31:
I ran in to all the same issues here on 0.976 and RPi B+. I fully flashed an SD Card (no upgrade) and was unable to interact with Volumio2 at all. Hook it up to the display, and I see all the OOM errors on the node process, if I kill the volumio service before the system bricks, it will not die. The hotspot sort of comes up, but is incredibly unstable (likely due to everything else going on with node).

I clean flashed 0.963 then, to go backwards, and see if this fixes the problem. I experienced the exact same behaviour as 0.976: completely unusable and hard locks the RPi+ after some time.

I attempted to run the volumio node app from the commandline to catch any errors with:

`NODE_ENV=production node index.js`

However I just get a 100% CPU and memory consuming process that prints no output.

I then clean flashed 0.928, to go a further step backwards. Still broken, unusable RPi+, and in this image the hotspot does not even start. Happy to start digging if someone points where!

If anyone would like the URLs for the updates (as I had to guess for 0.963) these are the list:
* 0.976 http://updates.volumio.org/pi/volumio/0.976/volumio-0.976-2016-07-28-pi.img.zip
* 0.963 http://updates.volumio.org/pi/volumio/0.963/volumio-0.963-2016-07-25-pi.img.zip
* 0.928 http://updates.volumio.org/pi/volumio/0.928/volumio-0.928-2016-06-30-pi.img.zip

Tihmann, 2016-07-31:
I did an update from the RC2 Hotfix Version. After that I can't connect anymore

samtygier, 2016-07-31:
I am also seeing OOM with volumio-0.976-2016-07-28-pi.img on a Pi Model B with 512MB.

I can ssh in after boot, but OOMs when I try to load a the web interface.

JedS, 2016-07-31:
@vostok4 : I did a fresh install of 0.928 on my RPi B+ and install also XBMC.
Both working without any problem.
The OOM start as soon as I update to latest Volumio version.

skikirkwood, 2016-07-31:
I'm up in Tahoe for a long weekend, but when I get home on Tuesday I can attempt to hook up a display to my Pi B+, kill the node process before it causes trouble and then restart under the Webstorm debugger and step through the code to see where the issue starts.

volumio, 2016-08-01:
FOA thanks for all the feedbacks. Imho this is what happens: achilles heel of volumio2 is RAM usage, since we load the Os on Ram and therefore on 512mb devices we get really no room left for the system. Couple of guesses (from a greek beach, no terminals here ):
- try to kill volumio before it starts and enable swap (the system will be slow but should st least work)
- The first boot of volumio will take more resources, since it needs to re create folders structure. Also i suspect that the main issue is created when we resize the backrounds, as we use a very memory hugry node module. The solution could be either disable this function in the sppearance plugin or delete all the backgrounds in the backgrounds folder under /volumio/app/plugins/miscllanea/appearance/backrounds

volumio, 2016-08-01:
Let me know how it goes...

vostok4, 2016-08-01:
@volumio perfect advice! I flashed 0.976, booted, stopped volumio service and deleted the backgrounds in the folder mentioned, and now the device works perfectly. Hotspot came up, joined the wifi network, and it seems to work very smoothly. Seems to be the background processing is indeed causing the issues.

And one huge adjustment on all my info, this is not a B+ (I get confused with all of these toys lying around), but an old RPi 1 model B, with HiFiberry DAC. I guess it doesn't change much since it still exists in the 512MB RAM bracked.

volumio, 2016-08-01:
Great! Thx Alex for the pr, we will need to rework this plugin a bit but at least we know what does not work.  Thx every body for the feedbacks.. Could you please report this on the forum as well so people on those devices know what to do?

vostok4, 2016-08-01:
Done: https://volumio.org/forum/976-rpi-rpi-system-not-booting-responsive-fix-t4426.html

Tihmann, 2016-08-01:
As I don't have a HDMI display could somebody provide a working image for download ?

gik98, 2016-08-01:
@Tihmann take a look at pull request #617 . I think you should be able to get an image from it in a quite straightforward manner (follow the README guide on the root of this repo).
Otherway, I think you could also deflate the downloaded image, open the squashfs image, remove backgrounds and repack everything - take a look here: [http://unix.stackexchange.com/questions/80305/mounting-a-squashfs-filesystem-in-read-write](url)

Actually, haven't tried none of the two above - anyway, if you just need a stable&working Volumio, you could still get version 1.55 from its repo on sourceforge

samtygier, 2016-08-02:
@volumio You say that the volumio OS is loaded into RAM which is why there is little left on small devices. Is that the whole filesystem being copied into RAM at boot, or just the writable part of it (like is common on linux live CDs)?

apinner, 2016-08-02:
@samtygier The whole OS runs in RAM, this is to reduce read/writes to the SD card to prolong life.

samtygier, 2016-08-02:
Reads should not effect the life of the SD card and most of the OS does not get written to on a day to day basis. It would be nice to be able to disable this on low RAM devices.

JedS, 2016-08-02:
Totally agreed with @samtygier.
I added XBMC to Volumio and I need to add more RAM for GPU to be able to run it.
With the gpu_mem=196, I still got OOM problem (even after deleting the backgrounds files).
I can reduce this paramater but having an option to avoid copying the whole filesystem on RAM will be better.

apinner, 2016-08-02:
It won't happen, well i very much doubt it anyway. Volumio has been specifically designed to run this way. 

Volumio provides you a complete OS that has been designed for a specific purpose, i don't think its particularly fair to then expect it to accommodate all of the possible modifications that fall outside of its intended use.

kermitology, 2016-08-06:
But if it renders a lower spec Pi useless? It wasn't cheap to get my DAC to Canada. This has to be part of the release plan unless you straight up say it will only run on higher spec Pis.

apinner, 2016-08-08:
@kermitology I think you are misunderstanding. Volumio will run on all Pi's, we will optimise where we can, reduce file sizes, improve code etc until it works properly. We're not turning our backs on lower spec boards. The solution is not a bodge that means part of the OS has to stay on an SD card. 

I am saying its unfair to expect volumio to compromise itself to make room for other systems it was never designed to run alongside.

gik98, 2016-08-12:
I saw the pull request has been merged. Any news about a release?

Saiyato, 2016-08-14:
I just tried this solution on an RPiB (same as B+, since it's rev 2), it seems to be booting fine if I delete the backgrounds, but still has OOM when I replace them with the new (optimised) ones.

If someone could confirm this before release? :)

gik98, 2016-08-15:
Ok, I found a little spare time and tried something on B+
Haven't been able to boot with optimized background, nor with no background at all. Actually, it boots, but becomes unresponsive after about half a minute.
I was working in a headless environment, will further inquire with a screen attached to the rpi to catch some logs.
Maybe I did something wrong while packing the image, even though it seems to me a similar behavior to the original image.

Saiyato, 2016-08-15:
I have had around 20 successful installations using the workaround provided, mostly due to the fact that I want a new MPD version to work with (to no avail so far).

The steps I have taken are as follows:

1. Create the SD card (download image 976 and write it to the card)
2. Boot up with an HDMI screen attached
3. As soon as I can login I do so
4. While the pi is still preparing the commandline I start typing the command (to minimize loss of time) 'systemctl stop volumio' I press enter afterwards using a USB keyboard
5. I connect to the pi using WinSCP and remove the background files
6. I use the command 'systemctl start volumio' to have it prepare volumio
7. I wait a minute or two and configure volumio, after which I can add one background (haven't tried more) and the pi (B for prepping and B+ for operation) shows no signs of OOM problems.

Since I was experimenting with new MPD versions the longest operational run was 4 days (without problems though). There is some instability when quickly switching between webradio and spotify, I think spopd dies.... But since the priority still is MPD I haven't looked into that yet.

I have not tried, nor will I in the near future, to re-pack the image. Mostly because I have no clue where to start and have little time to read into this the next couple of weeks. If I need to test something please let me know.

volumio, 2016-08-15:
I'm just working on a fix for the RAM issue, I am building now. I may ask you to try this new image... I'll keep you posted

volumio, 2016-08-15:
Ok, we have a fix for pi1: 
http://updates.volumio.org/pi/volumio/0.978/volumio-0.978-2016-08-16-pi.img.zip

This image works flawlessly (even if a bit slow on PI with 512MB of RAM). Basically, I added a script that checks if we have less than 530 MB of RAM, and if that's the case, creates a swap file and enables swap over there. The only culprit is that the very first boot process takes at least 5 minutes. 
But after that everything seems to be running smoothly. Let me know how it goes!

skikirkwood, 2016-08-16:
Booted up perfectly on my Pi B+ - including auto-detection/config of the IQAudio Pi-DAC+!

Installing the Spotify plugin and doing a thorough test now.

skikirkwood, 2016-08-16:
The automatic WiFi Hotspot is cool, but on the networking page I'm not seeing any of my wireless networks, hence I can't connect the Pi to the Internet.  Am I not understanding something?  I connected my Mac's WiFi to the Volumio hotspot and the UI came up fine, but then going to the Networking page my two wireless networks are not listed.

Tonymac32, 2016-08-16:
@skikirkwood I've had the wifi issue with the Panda wireless adapters, I swapped with the Edimax and had no problem.  Once I had the network set up with the Edimax I could swap in the Panda with no issue.  I'm guessing other adapters have a similar issue.

for @volumio , I'm having the same issue on a Pi Zero (976), dowloading the new image now.  For reference, for whatever reason, deleting the backgrounds *did not* solve the issue in the Pi Zero with IQAudio DACZero, it continued to throw out of memory errors.

Saiyato, 2016-08-16:
Will be testing the new image after work, will post back results.

gik98, 2016-08-16:
Yesss - it's working.

volumio, 2016-08-16:
@skikirkwood some wireless adapters cannot scan while in hotspot mode... The edimax does it just fine. In any case, you can configure your network by entering manually your SSD and PW

heis2201, 2016-08-16:
I can confirm that new image *volumio-0.978-2016-08-16-pi.img* is booting fine on Rpi1B+ with 512 MB.
Network issue persists: `/etc/network/interfaces` have to set `wlan0` to `dhcp`, UI keeps it on `manual`

volumio, 2016-08-16:
Does it mean you can't connect to wi-fi without changing to dhcp?

heis2201, 2016-08-16:
@volumio : Correct, I have to manually change `manual` to `dhcp`, otherwise no wifi connection possible.
I think, topic is covered here: #580

heis2201, 2016-08-16:
volumio-0.978-2016-08-16-pi used to work fine for a couple of hours, and now, suddenly it turned on the wifi hotspot and lost connection my home network. Why?

I really wonder why hotspot is enabled/installed by default. Isn't it convenient to connect Volumio to a wifi router that has a internet connection? (radio streaming, lan shares and so on). All devices that control my volumio and provide music are connected to the same home network anyways. So connection to that home network is mandatory. I would like to see this hotspot disabled by default. I really don't see the need for it at all, e.g. you have friends at home that want to stream something from their mobile device? Or control volumio? Give them access to your wifi home network! 

What's currently the best way to uninstall hostapd and thus free some RAM?

skikirkwood, 2016-08-16:
@volumio, where do I plug in my wireless network's SSD and PW?  I took a look at /etc/network/interfaces and the file was empty.  On the Network page of Volumio I don't see where I can manually enter my Wifi settings - see attached screenshot.  
[image]

Saiyato, 2016-08-16:
@volumio First impression is awesome! If anything it seems a bit faster than build 976 imho. Will move to the other pi (with a DAC) in an hour or so, see how it performs there. I'm loading (indexing into MPD) my music now (~5k songs).

First boot however was indeed slower, but that's only once, so no biggy.

skikirkwood, 2016-08-16:
Ok, I got 978 and my B+ with a Ralink RT5370 Wifi adapter on my wireless network.  Tried a bunch of things but finally copied over /etc/wpa_supplicant/wpa_supplicant.conf from my Rpi 3 and rebooted, which seemed to kill the hotspot.  Cool seeing the Multiroom Devices come up.
[image]

Tonymac32, 2016-08-17:
978 continuing to run with no issues (Pi Zero/IQAudio DacZero and HP Amp).  Some contact cleaner fixed my no audio issue, must have been crap built up in the flux from soldering

nosferatujr, 2016-08-17:
Just install in a Rasp Zero, and seams works fine with a wired network, but I can't get the wifi ok, always still 
[image]

when I tried to change from the UI the wpa_supplicant reflects the change:
ctrl_interface=/var/run/wpa_supplicant
network={
scan_ssid=1
ssid="Eternity_secure"
psk="securxxxxxxx"
}
but even whe I restart nothing change.

BTW  :
Searching plugins tab produce an empty result, 
Is any repository to download the plugins? (Spotify specially) Thanks

nosferatujr, 2016-08-17:
/etc/network/interfaces is empty. I confirm.

volumio, 2016-08-17:
Wait, which dongle are you using?

nosferatujr, 2016-08-17:
TP-LINK TL-WN725N

JedS, 2016-08-17:
@Volumio : does your script can be applyed to an already installed 0.976 version or do I need to flash with the new image ?

nosferatujr, 2016-08-17:
I start from the new image  volumio-0.978-2016-08-16-pi.img

makru83, 2016-08-17:
Using the latest Build here 2016-08-16 my Pi B+ is now booting fine, but I am not able to set a static IP Address for my wired Connection. When I set a static IP and reboot, I can see in the command line that while loading the wireless interface, it shows an error loading the dhcpd

[3 more replies]

closed 2017-06-13 by volumio