volumio/Volumio2 #600: v0.976 Playback options problems

chsims1, 2016-07-29:
Updated to v0.976 on RPi Zero, and there seems to a problem on the Playback options:

[image]

paviro, 2016-07-29:
Same on Pi 2 :)

xipmix, 2016-11-29:
Is this still a problem for you with 2.031? I have seen this on a pi2 with that version and it went away when I clicked on the selector next to the 'value not found' message. I recall it appeared after I changed the Mixer Type from hardware to none, but I can't reproduce it now.

macmpi, 2016-11-29:
Looks duplicate of [this](https://github.com/volumio/Volumio2-UI/issues/207): was fixed since (fresh) 2.01.

chsims1, 2016-11-29:
Not a problem anymore  for me on Pi0, Pi2 or Pi3.

closed 2017-06-02 by volumio