volumio/Volumio2 #601: Player control overlaps title of currently playing song

paviro, 2016-07-29:
[image]

apinner, 2016-07-29:
On what resolution? It should have switched to the mobile UI by that point.

paviro, 2016-07-29:
962px wide

paviro, 2016-07-29:
Using a radio stream. Titles sometimes get a bit longer switching happens long after that at about 770 pixels wide.

apinner, 2016-07-29:
What device is that?

paviro, 2016-07-29:
MacBook Pro 15"

apinner, 2016-07-29:
Haha, why are you running it at that resolution?

We can tweak it so it hides at lower resolutions, but it wont be a priority issue.

paviro, 2016-07-29:
Why should I not? Dont want a fullscreen chrome and I think it should definitely work at that resolution! 
[image]

apinner, 2016-08-01:
I didn't say you shouldn't, It's just an unusual resolution. 

Obviously this could be an issue at any resolution if the title is long enough. As i said, will think about a solution but its not going to be very high priority at the moment as it doesn't effect functionality.

paviro, 2016-08-01:
Alright perfect! :)

closed 2017-06-02 by volumio