volumio/Volumio2 #602: Web server for covers on mpdroid dont work on Volumio 2

Solaar74, 2016-07-29:
I was not remember whether the engine was installed in version 1.55, but it is not in version 2 and all operations of the topic
https://volumio.org/forum/web-server-for-covers-mpdroid-t456.html 
does not help.
[code]mkdir /var/www
chmod -R 777 /var/www/
ln -s /lib/mpd/music /var/www/covers[/code]
I had to install nginx. I read the instructions in English and Russian languages. Everything seems to be simple. But it was not possible to configure /etc/nginx/sites-available/default.local file correctly.  :?
MPDroid[spoiler][img]http://uploads.tapatalk-cdn.com/20160729/88923104aeee699147384bf74b96ab68.jpg[/img][/spoiler]
nginx[spoiler]a@ZOTAC ~ $ ssh volumio@192.168.1.3
volumio@192.168.1.3's password: 
                       ___                                      
                      /\_ \                        __           
         __  __    ___\//\ \    __  __    ___ ___ /\_\    ___   
        /\ \/\ \  / __`\\ \ \  /\ \/\ \ /' __` __`\/\ \  / __`\ 
        \ \ \_/ |/\ \L\ \\_\ \_\ \ \_\ \/\ \/\ \/\ \ \ \/\ \L\ \
         \ \___/ \ \____//\____\\ \____/\ \_\ \_\ \_\ \_\ \____/
          \/__/   \/___/ \/____/ \/___/  \/_/\/_/\/_/\/_/\/___/ 
        
             Free Audiophile Linux Music Player - Version 2.0

          C 2015 Michelangelo Guarise - Volumio Team - Volumio.org
                               

Volumio Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
volumio@volumio:~$ sudo systemctl enable nginx
Synchronizing state for nginx.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d nginx defaults
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "uk_UA.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
Executing /usr/sbin/update-rc.d nginx enable
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "uk_UA.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
volumio@volumio:~$ sudo /etc/init.d/nginx reload 
[sudo] password for volumio: 
[....] Reloading nginx configuration (via systemctl): nginx.serviceJob for nginx.service failed. See 'systemctl status nginx.service' and 'journalctl -xn' for details.
 failed!
volumio@volumio:~$ journalctl -xn
No journal files were found.
volumio@volumio:~$ systemctl -l status nginx.service
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled)
   Active: failed (Result: exit-code) since Thu 2016-07-28 10:39:20 UTC; 24h ago

Jul 28 10:39:17 volumio nginx[551]: nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
Jul 28 10:39:18 volumio nginx[551]: 2016/07/28 10:39:17 [emerg] 551#0: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
Jul 28 10:39:18 volumio nginx[551]: nginx: configuration file /etc/nginx/nginx.conf test failed
Jul 28 10:39:19 volumio systemd[1]: nginx.service: control process exited, code=exited status=1
Jul 28 10:39:20 volumio systemd[1]: Failed to start A high performance web server and a reverse proxy server.
Jul 28 10:39:20 volumio systemd[1]: Unit nginx.service entered failed state.
Jul 29 10:01:00 volumio systemd[1]: Unit nginx.service cannot be reloaded because it is inactive.
Jul 29 10:47:00 volumio systemd[1]: Unit nginx.service cannot be reloaded because it is inactive.
volumio@volumio:~$ sudo /etc/init.d/nginx restart
[ ok ] Restarting nginx (via systemctl): nginx.service.
volumio@volumio:~$ systemctl -l status nginx.service
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled)
   Active: active (running) since Fri 2016-07-29 11:18:25 UTC; 38s ago
  Process: 19047 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
  Process: 19040 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
 Main PID: 19049 (nginx)
   CGroup: /system.slice/nginx.service
           ├─19049 nginx: master process /usr/sbin/nginx -g daemon on; master_process on
           └─19050 nginx: worker process                           

Jul 29 11:18:25 volumio systemd[1]: Started A high performance web server and a reverse proxy server.
volumio@volumio:~$[/spoiler]

apinner, 2016-08-01:
Sorry, what is that supposed to be?

As i said on the forum, i'm not sure whether you are asking for help or reporting an issue?

Solaar74, 2016-08-01:
In version 1.55 was initially installed "nginx". In version 2.0, it is not. I sit and wait until something comes to replace him or to install "nginx"? Maybe "ighttpd"? I'm just asking you to help with "nginx" properly configured. Sorry for my Google translate )

Solaar74, 2016-08-01:
And bugs a lot. Do not adjust the volume, the library in the web interface and MPDroid differ in content, at a hotspot have ftp access and LAN - with the web interface disappeared after a reboot does not resume playback after every power necessary to rescan the library is not possible simply dragging and dropping files. pls in .m3u and WebRadio folder to add my radio station, after the addition of a web radio interface does not play .... Everything is beautiful, but does not work.(

closed 2016-11-20 by volumio