volumio/Volumio2 #603: Volumio 2 BeagleBone Black - kernel version?

codewarriorFX, 2016-07-29:
Hello - Do you know which kernel version will have the Volumio 2 for Beaglebone Black?

gkkpch, 2016-07-29:
There are no build scripts for Beaglebone(s) yet. As we use the overlayfs filesystem, the kernel  version must be at least 3.18. This means for Beaglebone(s) either 4.1 or 4.4, whichever one is considered stable.

codewarriorFX, 2016-07-29:
Ok, would it be possible if you can include these framebuffer tft drivers compiled on the next release for the beaglebone?

https://github.com/notro/fbtft

gkkpch, 2016-07-30:
fair question, but i don't think it is the teams policy to add any possible driver to the kernel. an alternative would be to build, whatever you intend to do with it, into a plugin. a. others can use it too b. we do not have to blow up the kernel size with drivers which a lot of others don't need.

apinner, 2016-08-01:
Agree, that should be built into a plugin.

codewarriorFX, 2016-08-01:
Sounds good to be a plugin.

When available, please let me know what kernel version for the beaglebone will be used, so we can build the framebuffer drivers for that specific version. For the hardware, I have a SSD1306, SSD1305 and ILI9341 displays for testing.

codewarriorFX, 2016-08-04:
Does the volumio build scripts have the option to add/configure the Kconfig and make files on the drivers before building?

gkkpch, 2016-08-04:
No, currently not because building the kernel is not part of the build procedure. There are also differences on how the kernel is integrated. Some devices install a package under chroot (Pi) some copy offline kernel files including u-boot to the image (odroid). I think it is feasible to add an optional script before the copying the kernel files, but this has not been done yet. That script should cover the kernel configuration and building. Would you be able to help with that?

codewarriorFX, 2016-08-16:
Optional script you mean that the user can choose which drivers to build?

gkkpch, 2016-08-16:
no, with optional I mean the build script could be extended to call a kernel building script. What you want to put in it would your choice. Not sure how much sense that really makes. I you want to build your own kernel, you can do that now anyway and use the build script like we do for the Odroids. Those ones pull the kernel and modules from a separate repo...

closed 2017-06-02 by volumio