volumio/Volumio2 #606: Creating Playlists from Music Library issues (Internal and Nas) Version 0.976

SonosKiller, 2016-07-29:
Adding a entire album or a catalouge to a playlist don't work 
Adding single songs works fine

apinner, 2016-08-01:
I was sure that i had done this successfully. Will investigate.

SonosKiller, 2016-08-01:
A quick test again !
From Music Library , Adding a entire album or a catalouge to a playlist don't work 
On 0.976 Pi  and 0.976 x86

Works fine on RC1 Pi (have no RC2 Pi left to test !)

apinner, 2016-08-01:
Thanks for testing again.

xipmix, 2016-11-29:
This may be fixed now.
I tested this on 2.031. Browse->Album, righthandmenu -> add to playlist
Worked both when adding to an existing playlist and a new playlist.
Could you try on your setup with 2.031 ?

SonosKiller, 2016-11-29:
Hi @xipmix
This was an old issue ! long before Album and Artist choices in the Browse menu  , and according to my "New version" test schedule and as far I remeber everything from Album and Artist works fine

My current issues when it comes to  creating and playing playlists is 
https://github.com/volumio/Volumio2/issues/854
https://github.com/volumio/Volumio2/issues/853

Thanks for everything

closed 2016-11-29 by SonosKiller