volumio/Volumio2 #607: "Clear and play" issues from Playlists and Music Library dialogue Version 0.976

SonosKiller, 2016-07-29:
Clear and play don't work , nothings happens , nothing ends up in queue and nothing plays, 
after using Clear and play the interface don't act properly, after a reload of the page it works properly again 

Note: Play and Add to queue works fine

apinner, 2016-08-01:
Why? Because we have had people reporting conflicting issues. I agree that it is the same as Play, but for some it doesn't seem to work.

https://github.com/volumio/Volumio2/issues/584 & https://github.com/volumio/Volumio2/issues/560

I think it needs further investigation.

SonosKiller, 2016-08-06:
Have to withdraw this issue !
Play , Clear and play , Add to queue works as intended !

I just wonder if any of my problems with e.g  Clear and play have something to do with that I have several Volumios installed with various versions and sometimes uses the Ip adress and sometimes uses volumio.local to access them ?
(the interface is different when I first access a Pi RC1 and then switch Multiroom device to a RC2 device)

closed 2016-08-06 by SonosKiller