volumio/Volumio2 #609: Can't control volume from MPD clients

jiri-novacek, 2016-07-30:
I can control volume from Volumio2 through WebUI, but volume can't be changed from MPD clients - Cantata (Windows) or MPDroid (Android). 
I can move volume sliders in those clients, but there is no hearable change and nothing happens with WebUI volume "knob". MPD clients are main "interface" I am using with Volumio, so as of now it's pretty difficult.
Using Volumio2 0.976 (didn't work in 0.928 either) on RPi B with USB "soundcard" (cheap shitty thing). 
Is there some setting which might help?

apinner, 2016-08-01:
That is because the volume control function in volumio is detached from mpd.

At some point in a future release an interface between the two will be added so it can be controlled by third party clients.

As this isn't an issue and we are aware of it, i am going to close.

closed 2016-08-01 by apinner

jiri-novacek, 2016-08-01:
It's not issue that user can't control volume of something called Volumio? :)
Don't get me wrong please, I understand that you are aware of this problem and I do appreciate all the work going to Volumio2 development. Thank you for response to my "issue", looking forward to future releases.

apinner, 2016-08-01:
You can control the volume.. We built you a beautiful UI to do it! ;)