volumio/Volumio2 #610: [V0.9760] Spotify does not work with Dutch language pack

Coolechriz, 2016-07-30:
When I volumio install and setup the language to Dutch and install spotifyeverything goes well. When I want to login into Spotify it will crash volumio interface. Which I need to reinstall the entire image.

I have no errorlog because i don't even know how to get this

nosferatujr, 2016-07-31:
Happens the same with catalan, instead of reinstall the image, just edit 
 /data/configuration/miscellanea/appearance/config-json 
and put 'English ' and 'en' on the  beggining of the file.

I thing before that is somethin related to the issue #599 but is not, ar two differents things

Coolechriz, 2016-07-31:
@nosferatujr And how to get by this data? I don't know how to get there. I can only reach the public files from my SD card.
I'm new to raspberry and volumio.

nosferatujr, 2016-08-01:
I access and edit this file through SSH. On Windows you need a SSH program like putty , or if you use Linux direclly from the terminal.

apinner, 2016-08-01:
This will be because each plugin needs its own translation. As can be found here; https://github.com/volumio/volumio-plugins/tree/master/plugins/music_service/spotify/i18n

I was under the impression that if a language was missing then it should default to English. Clearly that isn't happening.

nosferatujr, 2016-08-01:
I confirm, 
so at the moment a quick fix is 

cd /data/plugins/music_service/spop/i18n/
cp strings_en.json strings_xx.json 

xx is the two caracters that identify your language (a lot are missing, only is avaiable english and italian

the catalan translation for this file instead of copy:
{
  "spotify_username":"Nom de usuari Spotify",
  "spotify_password":"Contrasenya Spotify",
  "high_bitrate":"Alta qualitat",
  "search_results":"Nombre de resultats",
  "plugins":"Last.fm",
  "last_fm_username":"Nom usuari Last.fm",
  "last_fm_password":"Contrasenya Last.fm",
  "SEARCH_SONGS_SECTION":"Temes Spotify",
  "SEARCH_ALBUMS_SECTION":"Albums Spotity",
  "SEARCH_ARTISTS_SECTION":"Artistes Spotify"

}

closed 2017-06-02 by volumio