volumio/Volumio2 #611: V0.9760] Volume buttons do not work or very bad

Coolechriz, 2016-07-30:
When I want to put up or down the volume using the volume buttons but they do just one step something. When I try this again nothing happens at all after a few seconds. But as soon as I want to do the opposite volume exactly he does as the double step.

This was not happen to me on RC2 Hotfix. Please make a fix or something.

apinner, 2016-08-01:
We made some changes to the volume slider in the mobile UI in an attempt to make it easier to use. 

We are aware of an issue with the slider if you move it too quickly, we released it anyway to gather feedback on the changes.

hzpz, 2016-08-26:
The volume controls seem pretty broken to me atm. As @Coolechriz already mentioned, most of the time I can only click the up/down buttons once. A second click does nothing. If they *do* work a second time, they tend to ignore the "One Click Volume Steps" setting.

I am using 0.979 on a Raspberry Pi 3 and access the UI using a Mac with Firefox 48.

messismore, 2016-08-27:
Is there a way to prevent IOS bringing up the 'select stuff' UI when holding down on the sliding knob?

closed 2017-06-02 by volumio