volumio/Volumio2 #612: V0.976 deactivating Hotspot, connect to my wlan router

rnois, 2016-07-31:
Hello all, 
I have two issues with my Volumio2 installation. I like to deactivate the wlan hotspot and I like to contact my wlan network at home.
For the first issue, I have no idea how to deactivate the hotspot?
For the secound issue, in in the Network config I see several wifi networks around me, but not my own fritz box? How can I contact my fritz box and integrate my volumio2 into my network without cable connection? The way "connect to hidden network" does not work...?
I'm a user, now a system admin, means I have no deep no ledge about linux or so...
Many thanks for the help...!!

René

chsims1, 2016-07-31:
Is your "Fritz box" an ac router? If so, does your wifi connection on volumio support ac?

rnois, 2016-07-31:
Hi chsims1,

is a standard fritz box 7390 DSL-Lan/Wlan router with ac+n wlan mode. I had with the raspberry and other software (xmbc) a connection... but with Volumio2 I see only other networks around (also some fritz boxes) but no my network, and it is 1m away. Maybe there are to many networks around so that the list on the Volumio2 setup page is to short and can not list up all networks...?
Thanks!

chsims1, 2016-07-31:
Check what mode your router is using to create the network.  It sounds as though it might be creating an 'ac' network and your volumio wifi is type 'n' or even 'g'

gkkpch, 2016-07-31:
Switching off the hotspot will be added soon, in the meantime "remove" the hostapd software by renaming `/etc/hostapd` to `/etc/hostapd.unused`

rnois, 2016-07-31:
Hi,

I connect a EDIMAX Wifi-USB-Stick to my Raspberry3 and it works... No Hotspot available and I see my Fritzbox and I can connect. Works as expected... but only with the USB Wifi Stick not with the internal Wlan Chipset.
Any idea?

apinner, 2016-08-01:
I seem to recall that @volumio said that it will work best with the Edimax chips at the moment. 

We are aware of the open wifi issue, some security will be implemented in future. This is pre-release so at the moment it is for you to try and provide feedback.

heis2201, 2016-08-16:
Is renaming `/etc/hostapd` currently the best option to disable the hotspot? I renamed it and since then I am not the hotspot but volumio is also not connecting to my router (as it used to).

Is there another way to disable hostapd?

Thanks

kiwiASRALL, 2016-08-25:
I think it is not the best option. After reboot, the dhcpd service doesn't work and no IP is assigned to wired interface (eth0 in my case). Then without keyboard and screen attached on the Pi, it is not a solution.

closed 2017-06-02 by volumio