volumio/Volumio2 #613: LIBRARY IMPORT ERROR ON 0.976

ruudvddries, 2016-07-31:
Hi all,

This morning I upgraded volumio From v1.55 to v0.976 (the latest version anyways). I Mount all NAS drives I was using before and updated the library whill listining to a radio stream. After some time, playback stopped, and my library was empty. During the library scan the number of tracks was increasing steadily. 

I have tried setting "ro,noatime,sec=ntlmv2". The musicfile are located on a synology DS414 running the latest DSM 6.0 release. A log file can be found here; https://www.dropbox.com/s/c072dny78yo66qh/volumio.log?dl=0

And onder v1.55 the library was always full. What should/can I do to resolve this?

Regards,

Ruud

gkkpch, 2016-07-31:
are you running wireless or on an eth0 connection?

Gé

ruudvddries, 2016-07-31:
Wired! I Like stable network connections....

Ruud

yanyingjie007, 2016-08-01:
same here, it show my NAS was 16M, and no list of songs artists after reboot, i connected to WIFI and cable at the same time(different LANs).

closed 2017-06-02 by volumio