volumio/Volumio2 #615: Selecting Hifiberry only works after manually setting Output Device

dragotin, 2016-07-31:
When configuring my Raspberry based volumio device to use Hifiberry AMP with volumio version 0.976 it did not work after selecting *DAC Model* to *Hifiberry Amp* and switching on *I2S DAC* on. 

After rebooting, it did not play. To make it work the *Output Device* had to be set to *Hifiberry DAC* manually.

I think the *Output Device* should be set automatically if the user selects to use the Hifiberry Amp.

fmslottje, 2016-07-31:
I have the same problem on 0.976. I have to set the output at every reboot

Coolechriz, 2016-07-31:
For me the same problem

iso400, 2016-08-07:
I found a work around. I think this only occurs when de Hifiberry is detected automatically. Re-select both _output device_ and _DAC Model_ manually and _Save_. After this the settings persist after a reboot.

kaffeeringe, 2016-08-15:
Same problem with Hifiberry DAC Plus - after reboot "Audio Jack" is selected. I have to change that to have Sound.

volumio, 2016-08-15:
Which PI model do youhave ?

Coolechriz, 2016-08-15:
For me is it à raspberry pi3
With the dame problem

messismore, 2016-08-16:
Also happens on a Zero with with a Hifiberry Amp.

dragotin, 2016-08-16:
I experienced that on a Raspi 2, not sure which Model atm

volumio, 2016-08-16:
How did you guys reboot? By removing power or hitting reboot on the UI?

Coolechriz, 2016-08-16:
Reboot or shutdown from the interface. Direct removing power is not so good for hard- and software.

nicosea, 2016-08-21:
I have same the problem with PI3 and Pi-dac+ IQaudIO.

nicosea, 2016-08-23:
I tried to do what iso400 suggested, I re-select both output device and DAC Model manually and Save. After this the settings persist after a reboot, but no sound, I need to reselect IQaudIO DAC anyhow to be able to listen to music.

Is there a way to disable atutomatic Sound Card detection? So I can set PI-DAC+ manually once and avoid this problem on next reboots.

volumio, 2016-08-23:
It's on top of my priority list to fix.

michaelecker, 2016-08-23:
Thanks Michelangelo for taking care of this! 

 I have the same issue with a Pi1 B and Hifiberry DIGI

volumio, 2016-08-24:
Fixed with https://github.com/volumio/Build/commit/06a21f308ffdfb4feb2a53e70eb94acd007ab39d

Basically the new kernel was loading the i2s module before the integrated pi audio. If you don't want for a new release to come out, just add 

snd_bcm2835

to /etc/modules

Also, please someone add this information to relevant forum threads, so other people in need will benefit from the solution...

nicosea, 2016-08-24:
It's working for me! Thanks!

robvangeloven, 2016-09-09:
I don't know if it's been fixed in a new build. But for me in 0.979 I still have this issue. Using a hifiberry digi+ and raspberry pi 2.

On reboot it seems ok (from software). Cold boot is a 50/50 which gets selected, jack or digi

closed 2017-06-02 by volumio