volumio/Volumio2 #618: 2.0 RC2 path missing "mnt" in playlist

ruehaak, 2016-08-01:
This is my setup: raspberry 3 with hifiberry amp and Volumio 2.0 RC2. The music I play is on a USB stick. Almost everything works pretty smoothly except from playlists. When I add a song from my library that I haven't played before, the artwok is missing and I cannot play it. If I played the song directly from the library before, it works. Both the music file and artwork have different paths in the playlist. The ones that work correctly start with "mnt" the ones that don't directly with "USB":

This one works:
```{
"service": "mpd",
"uri": "mnt/USB/EERIE MUSIC/Acid Pauli/mst/03 palomitastep.mp3",
"title": "palomitastep",
"artist": "Acid Pauli",
"album": "mst",
"albumart": "/albumart?web=Acid%20Pauli/mst/large&path=mnt%2FUSB%2FEERIE%20MUSIC%2FAcid%20Pauli%2Fmst%2F03%20palomitastep.mp3"
}```

This one doesn't:
```{
"service": "mpd",
"uri": "/USB/EERIE MUSIC/Acid House Kings/Music Sounds Better With You/04 Under Water.mp3",
"title": "Under Water",
"artist": "Acid House Kings",
"album": "Music Sounds Better With You",
"albumart": "/albumart?web=Acid%20House%20Kings/Music%20Sounds%20Better%20With%20You/large&path=USB%2FEERIE%20MUSIC%2FAcid%20House%20Kings%2FMusic%20Sounds%20Better%20With%20You%2F04%20Under%20Water.mp3"
}```

I suspect it has to do with mounting the USB stick and I'd be glad for any assistance.

xipmix, 2016-08-01:
This looks similar to #509, which was related to NAS units.

xipmix, 2016-10-20:
pretty sure this is fixed, can you try with [2.001](http://updates.volumio.org/pi/volumio/2.001/volumio-2.001-2016-10-15-pi.img.zip)?

closed 2017-06-02 by volumio