volumio/Volumio2 #619: [0.976] Play / Pause button when playing webradio

Tihmann, 2016-08-01:
When playing webradio I still get problems correctly displaying the play/pause button (see https://github.com/volumio/Volumio2/issues/539). The error is not done completely. Now when you start a webradio stream the play / pause button changes from play to pause, but when you press pause it stops the webstream, but the button doesn't change from pause to play again, so you can't start the webstream again - you have to reload the page, then you get the play button. When you then press the play button the play button the webstream starts again, but can't ever stop it because you don't get the pause button to display again - even if you reload the page...
This still needs investigation

apinner, 2016-08-01:
Hmmmm... I wonder if this is because we are doing the same thing in both the back end and the UI.

https://github.com/volumio/Volumio2/blob/f0d4eae53d12e730a11f3a817845a14ebcc3ddad/app/statemachine.js#L145

https://github.com/volumio/Volumio2-UI/blob/master/src/app/themes/volumio/components/player-buttons/volumio-player-buttons.html#L27

Tihmann, 2016-08-01:
Maybe I was still on 0.928, I have flashed so many versions because of the OOM error...I will report back later

Tihmann, 2016-08-01:
So I verified the behaviour on 0.976. The bug is still there. When playing webradio I still get problems correctly displaying the play/pause button (see #539). The error is not done completely. Now when you start a webradio stream the play / pause button changes from play to stop, but when you press stop it stops the webstream, but the button doesn't change from stop to play again, so you can't start the webstream again - you have to reload the page, then you get the play button.

closed 2017-06-02 by volumio