volumio/Volumio2 #620: Very slow behavior via UPNP Renderer from Jriver (Volumio 2 RC2fix Odroid)

gabeg2, 2016-08-01:
If a song is playing, when i move to another song the player is very responsive in stopping the current song but the player takes roughly 10 seconds in loading the new track. 

Any thoughts? My other renderers are very responsive with jriver.

Thoughts?

gkkpch, 2016-08-01:
this could be a non C2-specific issue. I don't have jriver to test with, I need to find something to verify this with a Pi.

gabeg2, 2016-08-01:
I know volumio1 works fine on pi but will check volumio2 tonight.

gabeg2, 2016-08-03:
jriver works fine with volumio2 on pi.

gabeg2, 2016-08-09:
ok.  i tried the pc version and i get the same slow behavior with volumio2 as I did on the odroid 2.

volumio, 2016-08-09:
It may depend on the buffer size, try decreasing it.

gabeg2, 2016-08-09:
Ok.   That seems counter-intuitive but will try.

DaveGahan, 2017-03-29:
Hi,

Same issue here with my setup :

Bubbleupnp (Android Nougat) -> Volumio 2 (PI 3) <- Synology Nas (minidlna)

- WebUI  : works flawlessly (max 1 second to play the first or next song)
- MPDroid  : as well as WebUI
- Bubbleupnp : takes about 6/7 seconds to play first or next song

I tried to decrease buffer size but it has no effect.

I didn't have this problem when I was using Volumio 1.

Any idea ?

Thanks.

closed 2017-06-02 by volumio