volumio/Volumio2 #622: [0.976] Fourth webradio in list

Tihmann, 2016-08-01:
When adding four webradios, you are not able to see the complete menu the upper bit is truncated
[image]

apinner, 2016-08-02:
Confirmed, looks like an overflow issue. 

Will look at a fix.

SonosKiller, 2016-08-07:
Almost the same effect on the fourth item in a Playlist,in a Spotify Playlist, Favorites, and  Album list (missing or a half play button)

bst27, 2016-10-31:
To add some additional info about this issue: 

I can confirm a similiar behaviour in Google Chrome and Safari on an iPhone 5S with Volumio 2.001. The first items menu is created correctly (image 1) but the menu of the second item is truncated (image 2).

[image]
[image]

closed 2017-06-02 by volumio