volumio/Volumio2 #623: [0.976] Volumio crashes when playing Klassikradio

Tihmann, 2016-08-01:
When adding the webstation Klassikradio http://stream.klassikradio.de/live/mp3-128/stream.klassikradio.de/play.pls and playing it from the menu nothing happens - no playback. But after doing so, you are not able to play another webradio stream, you have to restart. No error is display in the GUI

apinner, 2016-08-02:
This is probably an mpd issue and requires a later version of mpd

janpieterz, 2016-08-11:
I have odd radio behavior with a radio (1920's network) that works fine sometimes, but sometimes blocks up the whole PI. Don't know if it helps or is related, but if so, this is the url: `http://tess.fast-serv.com:8570/listen.pls`

kuchendieb, 2016-08-11:
I see the same thing on my RPI. After trying to play kassik radio and some other channels volumio wont play any webradio. You have to restart

tim-felix-lange, 2016-09-03:
Same for me with http://listen.technobase.fm/dsl.pls, after starting, it doesn't play and won't play any other stream until I restart. Looks like some problem with .pls files.

xipmix, 2017-03-13:
I tested this on 2.118 (rpi2) and klassikradio plays ok. There is a complaint in the log about the file:
```
mpd[596]: pls: Invalid PLS file: 'Key file does not have key 'NumberOfEntries''
```
but it plays fine. After playing that url I can play another with no issue (it was an .m3u though).
These also played ok, with no reboots required:
 * http://tess.fast-serv.com:8570/listen.pls
 *  http://listen.technobase.fm/dsl.pls

I think this one is fixed, please let us know if it's not working for you.

xipmix, 2017-04-06:
If you think it is fixed, please could you close the issue.

closed 2017-06-02 by volumio