volumio/Volumio2 #626: stutter sound after latest OTA update

andreasruedel, 2016-08-03:
greetings
after reinstalling using RC2 hotfix image I set up volumio2 again and under "system" I had "upgrade available" (over the air) - installed it and rebooted. since then I get stutters/choppy sound when playing back mp3 files (mostly 320kbps ones) - it's pretty random and when checking the console using 'top' I could not see any high / hung things. (load was 0.3 at max)
using raspberry pi 2 with hifiberry dacplus 
can you advice how I should troubleshoot/report this? 
thanks
Andreas

apinner, 2016-08-03:
Which device?

andreasruedel, 2016-08-03:
raspberry pi 2 with hifiberry dacplus

Bubbamike, 2016-08-19:
I have the same issue. Raspberry Pi2 via USB to a PS Audio NuWave DAC. I've fooled around with moving cables, and buffer sizes but the issue is still there.

Bubbamike, 2016-08-21:
[Volumio problem.zip](https://github.com/volumio/Volumio2/files/428638/Volumio.problem.zip)
Here's an example of the problem.

volumio, 2016-08-21:
Wow that's an issue indeed... I really don't have a clue on how to solve this... Could you try changing the output device and then subsequently the mixer?

Sander884, 2016-10-09:
Same problem overhere with Raspberry Pi 2 Model B and HifiBerry Digi+ (digital output).
Playing .flac from smb share and also internet radio :(

closed 2017-06-02 by volumio