volumio/Volumio2 #629: VOLUMIO v.0.976 x86 don't work on AMD platform ?

Solaar74, 2016-08-05:
VOLUMIO don't boot on Zotac Fusion E-350 motherboard. The blinking cursor. No access network.

balbuze, 2016-08-05:
I tested it on a hp pavillon dm1 with a amd e350 and it works, but hotspot not working yet. Wifi works after configuration.

Solaar74, 2016-08-05:
Where is the problem?

balbuze, 2016-08-05:
The hostapd problem seems to come from a missing line in config file for the build. It's beeing corrected and will be tested. Gkkpch is working on it

Solaar74, 2016-08-05:
OK. Will wait. I wonder how the system works on the PC.)

gkkpch, 2016-08-05:
i'm already working on this. similar to @balbuze medion problem. @Solaar74 can download a special version as per PM. I think I know where it stops, but have no clue why. 
hostapd is a step further, visible and connectable. Will be in the next release...

gkkpch, 2016-08-06:
@Solaar74 you have a PM

gkkpch, 2016-08-11:
i did find an issue and resolved it with the help of solaar74. Unfortunately, it left us with the next one. After successfully starting volumio 2 on a device that did not work before, there is an issue with reboot. Trying to find out what actually happens, I don't see any relation to the initial issue yet.

closed 2017-06-02 by volumio