volumio/Volumio2 #630: Volumio crashes when connected to a network without internet connection

paviro, 2016-08-08:
When you connect Volumio to a network without internet connection it will crash show the re connection spinner and be available again shortly after. 

Latest Volumio version available via OTA.

apinner, 2016-08-08:
What makes you think its crashing?

paviro, 2016-08-08:
It shows the reconnection spinner than the UI then the spinner again shortly after. In that time to startup sound comes again and the spinner goes away.

apinner, 2016-08-08:
OK, so its more than just the spinner. I thought it may have just been renewing its IP address and the network temporarily went down.

Anything in the logs

paviro, 2016-08-08:
I do not know. I have internet now so I can't test it anymore.

apinner, 2016-08-08:
Huh? Are you saying your internet went down? 

I think you might be giving us a bit of a red herring here..

paviro, 2016-08-08:
No I moved to another city setup the network and connected volumio to it. But I did not yet have internet just got it today and until I had real internet and not just a local network the things I described happened, as soon as internet was available it worked again as always.

scrossi, 2016-08-17:
Not a red herring. I have had daily experience with the same problem on two PI B+.

xipmix, 2017-03-13:
If this still occurs in 2.118, could you send in a debug log?
 * go to the /dev url in volumio
 * fill in the "Send Log or bug report" item, referencing issue 630.
 * hit the Send button.

If it's fixed, please let us know that.

chsims1, 2017-03-13:
Recent post on forum [https://volumio.org/forum/volumio2-031-can-select-tracks-play-t5608.html](url) (see second page)

xipmix, 2017-03-14:
Doh! 'Send the log report' ... when there's no internet available. Double-doh! 
If you want to try, this might work better:
 * shut down volumio and start up again.
 * log in to the volumio host, and run this script (which does the same things as the log report)
```
#!/bin/sh
echo "Logging to /tmp/report.log"
exec 1>/tmp/report.log 2>&1
cat /proc/version
cat /etc/os-release
ifconfig
iwconfig
aplay -l
ps -ef
journalctl -p 7
```
like so:
```
volumio@volumio:~$ sudo sh report.sh
volumio@volumio:~$ gzip /tmp/report.log
```
Then you will have to send report.log.gz to a machine where you can save it to a usb stick and when you have internet, upload.

Gridgway1960, 2017-03-14:
Does this help...

Volumio Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
volumio@mansfield2:~$ sudo journalctl -f
-- Logs begin at Sun 2017-02-19 15:53:56 UTC. --
Feb 19 16:04:40 mansfield2 sshd[1008]: lastlog_openseek: Couldn't stat /var/log/ lastlog: No such file or directory
Feb 19 16:04:40 mansfield2 sshd[1008]: lastlog_openseek: Couldn't stat /var/log/ lastlog: No such file or directory
Feb 19 16:04:44 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/%252 2Weird%2520Al%2522%2520Yankovic/%2522Weird%2520Al%2522%2520Yankovic/info.json'
Feb 19 16:04:46 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/4her o/4hero/info.json'
Feb 19 16:05:41 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/5%25 20Seconds%2520of%2520Summer/5%2520Seconds%2520of%2520Summer/info.json'
Feb 19 16:05:42 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/7%25 20Second%2520Of%2520Love/7%2520Second%2520Of%2520Love/info.json'
Feb 19 16:05:42 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/3%25 20Doors%2520Down/3%2520Doors%2520Down/info.json'
Feb 19 16:05:42 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/7%25 20Seconds%2520Of%2520Love/7%2520Seconds%2520Of%2520Love/info.json'
Feb 19 16:06:09 mansfield2 sudo[1035]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
Feb 19 16:06:09 mansfield2 sudo[1035]: pam_unix(sudo:session): session opened fo r user root by volumio(uid=0)
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::executeOnPlugin: appearance , getUiSettings
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::volumioGetState
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426523] CoreStateMachine::getState
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426525] CorePlayQueue::getTrack 0
Feb 19 16:07:06 mansfield2 volumio[800]: info:
Feb 19 16:07:06 mansfield2 volumio[800]: [1487520426526] ---------------------------- Client r equests Volumio state
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426527] Listing playlists
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Feb 19 16:07:06 mansfield2 volumio[800]: Plugin multiroom or method getMultiroom not found
Feb 19 16:07:06 mansfield2 volumio[800]: info:
Feb 19 16:07:06 mansfield2 volumio[800]: [1487520426543] ---------------------------- Client r equests Menu Items
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::volumioGetBrowseSources
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::volumioGetQueue
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426548] CoreStateMachine::getQueue
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426548] CorePlayQueue::getQueue
Feb 19 16:07:06 mansfield2 volumio[800]: info:
Feb 19 16:07:06 mansfield2 volumio[800]: [1487520426549] ---------------------------- Client r equests Volumio queue
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426550] InterfaceWebUI::pushState
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426568] InterfaceWebUI::pushQueue
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426570] ------------------------------ 49ms
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426571] ------------------------------ 24ms
Feb 19 16:07:34 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/A%2520New%2520Funky%2520Generation%2520feat.%2520Joy%2520Rose/A%2520New%2520Funky%2520Generation%2520feat.%2520Joy%2520Rose/info.json'
Feb 19 16:07:35 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/A1/A1/info.json'
Feb 19 16:08:08 mansfield2 volumio[800]: info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
Feb 19 16:08:08 mansfield2 volumio[800]: info: CURURI: artists://
Feb 19 16:08:30 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/AC%252FDC/AC%252FDC/info.json'
Feb 19 16:08:30 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/Adamski/Adamski/info.json'
Feb 19 16:08:31 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/ABC/ABC/info.json'
Feb 19 16:08:31 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/ATB/ATB/info.json'
Feb 19 16:08:34 mansfield2 volumio[800]: info: CoreCommandRouter::executeOnPlugin: appearance , getUiSettings
Feb 19 16:08:34 mansfield2 volumio[800]: info: CoreCommandRouter::volumioGetState
Feb 19 16:08:34 mansfield2 volumio[800]: info: [1487520514694] CoreStateMachine::getState
[... cut ...]
volumio@mansfield2:~$ Feb 19 16:10:47 mansfield2 ntpd_intres[607]: host name not found: 2.debian.pool.ntp.org
-bash: Feb: command not found
volumio@mansfield2:~$ Feb 19 16:11:43 mansfield2 ntpd_intres[607]: host name not found: 3.debian.pool.ntp.org
-bash: Feb: command not found
volumio@mansfield2:~$ Feb 19 16:12:15 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/%2522Weird%2520Al%2522%2520Yankovic/%2522Weird%2520Al%2522%2520Yankovic/info.json'
-bash: Feb: command not found
volumio@mansfield2:~$

Gridgway1960, 2017-03-14:
Sorry, not sure what happened there with the cut and paste, is it ok?

xipmix, 2017-03-14:
It's the github markup language it interprets `~` as "start/stop strikethrough".
To paste without the markup mangling things, you need to wrap the text in three ` marks, see  https://help.github.com/articles/creating-and-highlighting-code-blocks

\`\`\`
for example
\`\`\`

Thanks for taking the time to send the log.                          
It doesn't show any crashing that I can see but it gives some pointers.

Gridgway1960, 2017-03-14:
```
Free Audiophile Linux Music Player - Version 2.0

C 2015 Michelangelo Guarise - Volumio Team - Volumio.org

Volumio Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
volumio@mansfield2:~$ sudo journalctl -f
-- Logs begin at Sun 2017-02-19 15:53:56 UTC. --
Feb 19 16:04:40 mansfield2 sshd[1008]: lastlog_openseek: Couldn't stat /var/log/ lastlog: No such file or directory
Feb 19 16:04:40 mansfield2 sshd[1008]: lastlog_openseek: Couldn't stat /var/log/ lastlog: No such file or directory
Feb 19 16:04:44 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/%252 2Weird%2520Al%2522%2520Yankovic/%2522Weird%2520Al%2522%2520Yankovic/info.json'
Feb 19 16:04:46 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/4her o/4hero/info.json'
Feb 19 16:05:41 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/5%25 20Seconds%2520of%2520Summer/5%2520Seconds%2520of%2520Summer/info.json'
Feb 19 16:05:42 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/7%25 20Second%2520Of%2520Love/7%2520Second%2520Of%2520Love/info.json'
Feb 19 16:05:42 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/3%25 20Doors%2520Down/3%2520Doors%2520Down/info.json'
Feb 19 16:05:42 mansfield2 volumio[800]: ERROR getting albumart: Got error: geta ddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/7%25 20Seconds%2520Of%2520Love/7%2520Seconds%2520Of%2520Love/info.json'
Feb 19 16:06:09 mansfield2 sudo[1035]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
Feb 19 16:06:09 mansfield2 sudo[1035]: pam_unix(sudo:session): session opened fo r user root by volumio(uid=0)
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::executeOnPlugin: appearance , getUiSettings
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::volumioGetState
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426523] CoreStateMachine::getState
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426525] CorePlayQueue::getTrack 0
Feb 19 16:07:06 mansfield2 volumio[800]: info:
Feb 19 16:07:06 mansfield2 volumio[800]: [1487520426526] ---------------------------- Client r equests Volumio state
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426527] Listing playlists
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Feb 19 16:07:06 mansfield2 volumio[800]: Plugin multiroom or method getMultiroom not found
Feb 19 16:07:06 mansfield2 volumio[800]: info:
Feb 19 16:07:06 mansfield2 volumio[800]: [1487520426543] ---------------------------- Client r equests Menu Items
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::volumioGetBrowseSources
Feb 19 16:07:06 mansfield2 volumio[800]: info: CoreCommandRouter::volumioGetQueue
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426548] CoreStateMachine::getQueue
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426548] CorePlayQueue::getQueue
Feb 19 16:07:06 mansfield2 volumio[800]: info:
Feb 19 16:07:06 mansfield2 volumio[800]: [1487520426549] ---------------------------- Client r equests Volumio queue
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426550] InterfaceWebUI::pushState
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426568] InterfaceWebUI::pushQueue
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426570] ------------------------------ 49ms
Feb 19 16:07:06 mansfield2 volumio[800]: info: [1487520426571] ------------------------------ 24ms
Feb 19 16:07:34 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/A%2520New%2520Funky%2520Generation%2520feat.%2520Joy%2520Rose/A%2520New%2520Funky%2520Generation%2520feat.%2520Joy%2520Rose/info.json'
Feb 19 16:07:35 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/A1/A1/info.json'
Feb 19 16:08:08 mansfield2 volumio[800]: info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
Feb 19 16:08:08 mansfield2 volumio[800]: info: CURURI: artists://
Feb 19 16:08:30 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/AC%252FDC/AC%252FDC/info.json'
Feb 19 16:08:30 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/Adamski/Adamski/info.json'
Feb 19 16:08:31 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/ABC/ABC/info.json'
Feb 19 16:08:31 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/ATB/ATB/info.json'
Feb 19 16:08:34 mansfield2 volumio[800]: info: CoreCommandRouter::executeOnPlugin: appearance , getUiSettings
[... cut ...]
volumio@mansfield2:~$ Feb 19 16:10:47 mansfield2 ntpd_intres[607]: host name not found: 2.debian.pool.ntp.org
-bash: Feb: command not found
volumio@mansfield2:~$ Feb 19 16:11:43 mansfield2 ntpd_intres[607]: host name not found: 3.debian.pool.ntp.org
-bash: Feb: command not found
volumio@mansfield2:~$ Feb 19 16:12:15 mansfield2 volumio[800]: ERROR getting albumart: Got error: getaddrinfo EAI_AGAIN ws.audioscrobbler.com:80 for Infopath '/data/albumart/web/%2522Weird%2520Al%2522%2520Yankovic/%2522Weird%2520Al%2522%2520Yankovic/info.json'
-bash: Feb: command not found
volumio@mansfield2:~$
```

Gridgway1960, 2017-03-14:
Cool! Thanks, much more readable.  Sorry for being such a newbie!

closed 2017-06-02 by volumio