volumio/Volumio2 #636: cue file kills Volumio

balbuze, 2016-08-13:
Can't read cue file, it kills Volumio.
last message is :  
< [1471116433330] ---------------------------- Client requests add and Play Volumio CUE entry
< info: CoreCommandRouter::executeOnPlugin: mpd , addPlayCue
< info: Error:
< ReferenceError: self is not defined
<     at ControllerMpd.addPlayCue (/volumio/app/plugins/music_service/mpd/index.js:139:64)
<     at CoreCommandRouter.executeOnPlugin (/volumio/app/index.js:407:29)
<     at Promise._successFn (/volumio/app/plugins/user_interface/websocket/index.js:176:33)
<     at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
<     at _combinedTickCallback (internal/process/next_tick.js:67:7)
<     at process._tickCallback (internal/process/next_tick.js:98:9)
< /volumio/node_modules/kew/kew.js:198
<       throw e
<       ^

< TypeError: self.getI18NString is not a function
<     at Promise._successFn (/volumio/app/plugins/user_interface/websocket/index.js:183:66)
<     at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
<     at _combinedTickCallback (internal/process/next_tick.js:67:7)
<     at process._tickCallback (internal/process/next_tick.js:98:9)
program terminated
debug>

lowrideris, 2016-09-19:
Same problem here on x86 build latest version

volumio, 2016-09-19:
Fixed, it will be on next release

closed 2016-09-19 by volumio