volumio/Volumio2 #640: RFE: Add "Reset" function to UI

skikirkwood, 2016-08-14:
It's possible to get Volumio in strange states that only rebooting will fix.  A nice alternative would be to have a "Reset" button on the System Settings page that would delete the queue, kill the node process, kill the mpd process and then when these automatically restarted many locked up situations would be fixed without a reboot.  They start up real fast too.

WalterBorgstein, 2016-08-15:
I agree. Being a novice user I sometime feel restarting is the only option. But maybe I'm just not familiar enough yet with Volumio. Love to participate though.

heis2201, 2016-08-17:
as far as I remember, there is a "reset to factory defaults" button under settings. That should do it.

skikirkwood, 2016-08-17:
I believe that would be too much.  I want my MPD database intact, WiFi settings, MyWebRadio favorites, installed plugins, etc.  So I only want to get a fresh start without changing persistent data.

volumio, 2016-09-28:
Just reload the page

closed 2016-09-28 by volumio