volumio/Volumio2 #641: NFS mount doesn't show up

kaffeeringe, 2016-08-15:
I mounted an NFS share from my Synology NAS. It shows up in "Settings>My Library". But I can't scan the collection and the share doesn't show up in Browser>My Library.

I looked into it via SSH. It shows up in the list for _mount -l_  - but when I try to _cd_ into it, I don't have the permissions.

Hardware: Raspi2+Hifiberry DAC Plus
Software: Volumio RC2 Hotfix - No Updates.

kaffeeringe, 2016-08-15:
I chmod the mount to 777 and now it also shows up in the library.

WalterBorgstein, 2016-08-15:
Will try this this week! Thanks.

kaffeeringe, 2016-08-15:
Anyway the frontend doesn't seem to start _mpc update_ when I try to rescan. I started it in the terminal.

WalterBorgstein, 2016-08-15:
How?

kaffeeringe, 2016-08-15:
Using: _mpc update_

But somehow the share seems to be empty. This is not yet the solution... :-/

WalterBorgstein, 2016-08-15:
Too bad :(. You would expect this to be a normal share to be used. I'd like to helping testing.

WalterBorgstein, 2016-08-20:
[image]

I do see connection activity on my Synology, but no update of the library.

volumio, 2016-08-20:
NFS is quite troublesome... Thanks for the help, unfortunately I'm out of clues right now..

WalterBorgstein, 2016-08-20:
It would be helpful if someone posts a successful mount to an Sinology NFS share. Including prints of the config on Sinology side and DF overview.

WalterBorgstein, 2016-08-20:
[image]
[image]

Whoera! Succes. I modified the config on the Synology NFS share. Now I can CD into the MNT under
/mnt/NAS/test and see the FLAC files.

Not quite sure I my files are save now. So I will be limiting access step by step to see where and when it fails.
The attachment shows the NFS share settings and the user permissions. Basically I gave everyone R/W.

WalterBorgstein, 2016-08-20:
[image]

Proof that files on the Synology have been added to the Library. I ran a 
`

> MPC Update NAS

` 

te be sure.

WalterBorgstein, 2016-08-20:
After a reset to Factory Mode. I tried to add the NFS via the GUI. Instantly it ran an update of the library. 
I've added the Volumio config and results also. 

**Conclusion**
Volumio is able to handle Synology NFS shares and automatically update the library. Challenge is in configuration of the NFS share on the Synology. I will try to limit the amount of access to see when and if the MNT breaks. 

[image]

[image]

volumio, 2016-08-20:
It would be great if you could find the optimal conf on Synology side and write a small guide for it... It will be very much appreciated on the forum ...

WalterBorgstein, 2016-08-20:
# Adding Music from a shared folder on a Synology
`tested on Synology DS412 in combination with Volumio 0.978 for Raspberry Pi 3`

Since both Synology and Volumio for Raspberry PI are Unix based the preference is to use NFS type file sharing.
> For more information visit: [https://en.wikipedia.org/wiki/Network_File_System](url)

**Synology preparation**
To enable NFS on Synology follow the detailed guide from Synology [https://www.synology.com/en-global/knowledgebase/DSM/tutorial/File_Sharing/How_to_access_files_on_Synology_NAS_within_the_local_network_NFS]

The guide will explain in great detail the steps required.

The final NFS rule configuration is shown in the screenshot. The most important part is the **Squash** setting. The required access is RW and since it's wise to disable the Guest account on your Synology, the Squash setting must be set to Map all users to admin. 

[image]

**Volumio preparation**

- Go to settings > My Music
- + Add New Drive
- Make sure you configure the share as shown in the screenshot.

> volume1 is normally the first part of the path, followed by the name of your share. In this case **Music**

Since we shared using NFS no username and password is required and

[image]

WalterBorgstein, 2016-08-20:
@kaffeeringe Could you give this a try? Preferably on a clean install.

volumio, 2016-08-20:
Straight on the docs...
https://volumio.github.io/docs/Good_to_Knows/Mounting_an_NFS_Share.html

madelgijs, 2016-08-30:
When using Volumio (0.979) with my Synology DS112+ PATH had to be: "**/**volume1/music" - without the trailing slash before volume1 it didn't mount.

WalterBorgstein, 2016-08-30:
Could you provide a print screen? That way @volumio can add that to the documentation.

exetico, 2016-10-01:
I did add a NFS-mount, with all my music. Actually i just wanted to add a mount with my Podcasts (Downloaded from a RSS-feed on my Synology NAS).

I removed the mount, and did try to add the Podcasts again.

Volumio version: 0.979 

But... The permissions isn't right. I did a lot of tries, with no luck at all.
`Oct 01 14:00:30 volumio mpd[594]: client: [6] opened from local
Oct 01 14:00:30 volumio mpd[594]: client: [6] closed
Oct 01 14:00:30 volumio volumio[552]: info: [1475330430196] ControllerMpd::sendMpdCommand status
Oct 01 14:00:30 volumio volumio[552]: info: Mpd Status Update: update
Oct 01 14:00:30 volumio volumio[552]: info:
Oct 01 14:00:30 volumio volumio[552]: [1475330430200] ---------------------------- MPD announces state update
Oct 01 14:00:30 volumio mpd[594]: errno: Failed to open 'NAS/Podcast': Permission denied
Oct 01 14:00:30 volumio volumio[552]: info: [1475330430202] sending command...
Oct 01 14:00:30 volumio volumio[552]: info: [1475330430203] ControllerMpd::getState
Oct 01 14:00:30 volumio volumio[552]: info: [1475330430204] ControllerMpd::sendMpdCommand status
Oct 01 14:00:30 volumio volumio[552]: info: [1475330430205] sending command...
Oct 01 14:00:30 volumio volumio[552]: info: [1475330430211] Database update started
Oct 01 14:00:30 volumio volumio[552]: COMMON.MY_MUSIC
Oct 01 14:00:30 volumio volumio[552]: [ 'COMMON', 'MY_MUSIC' ]
Oct 01 14:00:30 volumio volumio[552]: COMMON.SCAN_DB
Oct 01 14:00:30 volumio volumio[552]: [ 'COMMON', 'SCAN_DB' ]
Oct 01 14:00:30 volumio volumio[552]: info: [1475330430227] ControllerMpd::sendMpdCommand status
Oct 01 14:00:30 volumio volumio[552]: info: Mpd Status Update: update
Oct 01 14:00:30 volumio volumio[552]: info:
`
As you can see, the permisisons isn't right.

I did add the mount in the interface - but nothing helps:
`root@volumio:/mnt/NAS# ls -al
total 16
drwxrwxrwx 1 root root 4096 Oct  1 14:00 .
drwxrwxrwx 1 root root 4096 Aug 20 16:39 ..
d--------- 4 root root 4096 Oct  1 12:46 Podcast

Then trying to remove the mount, it also fails:
`
Oct 01 14:04:12 volumio volumio[552]: info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
Oct 01 14:04:15 volumio volumio[552]: info: CoreCommandRouter::executeOnPlugin: networkfs , deleteShare
Oct 01 14:04:15 volumio volumio[552]: error: Cannot Delete Folder. Error: Error: Command failed: rm -rf /mnt/NAS/Podcast
Oct 01 14:04:15 volumio volumio[552]: rm: cannot remove '/mnt/NAS/Podcast': Device or resource busy
Oct 01 14:04:16 volumio volumio[552]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 01 14:04:16 volumio volumio[552]: info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
Oct 01 14:04:20 volumio volumio[552]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
`
I did try with another name like "Podcasting", but nothing helps. 
[image]

closed 2017-06-02 by volumio