volumio/Volumio2 #642: Can't uninstall Spotify plugin

kaffeeringe, 2016-08-15:
There seems to be no way of uninstalling the plugin. But I still have the button "Install" - which fails and correctly tells me, that the plugin is already installed.

closed 2016-09-28 by volumio

wmoecke, 2018-01-28:
Yeah, I'm stuck with this plugin which doesn't work, doesn't uninstall and is just there occupying space. How do I manually get rid of this plugin?

lucasgut, 2018-02-14:
Same issue here I get an undefined  error when uninstalling  the plugin. Why has this issue been closed?

volumio, 2018-02-14:
Fixed https://github.com/volumio/Volumio2/commit/1b60c9a1a9e254ebaa7c37a77df0ea7738bdbe5a

macmpi, 2018-02-14:
While at it, wouldn't it make sense to also remove **plugin's configuration folder** during uninstall ([here for instance](https://github.com/volumio/Volumio2/blob/1b60c9a1a9e254ebaa7c37a77df0ea7738bdbe5a/app/pluginmanager.js#L1219))?
This would make sure one start from a "clean" state if it is re-installed later.

volumio, 2018-02-14:
Thanks for the suggestion, done https://github.com/volumio/Volumio2/commit/e512b5064eff470f696d69612fb11e859d020d92