volumio/Volumio2 #643: Spotify does not play back anything

kaffeeringe, 2016-08-15:
I can't listen to anything. I don't know if it's due to the fact that I only have a free account. I can encue tracks, but I can't hear them the Raspi remains silent.

skikirkwood, 2016-08-15:
You need a premium account to use the Spotify plugin on Volumio.

kaffeeringe, 2016-08-15:
Ah! Thanks!

skikirkwood, 2016-08-15:
After we fully migrate to the new Spotify API we'll be able to verify you have a paid account and issue an error otherwise, but our current integration doesn't have a method for us to check.

kaffeeringe, 2016-08-15:
Very good! But maybe you could also add a hint to the plugin's description. Instead of "Spotify Plugin for Volumio: Browse, Play and discover Spotify Musik" it could be: "Browse, Play and discover Spotify Musik. Premium Accounts only!" - I think it's clear, that if you install a plugin called "Spotify" in Volumio, that it's the "Spotify plugin for volumio" ;-)

skikirkwood, 2016-08-15:
Yes, we need to do that.  I think we did have the verbiage for premium accounts only in the previous version of Volumio.  

After the 2.0 release stabilizes we’ll be having you authenticate your account with the Spotify web site, and that will allow us to access much more of your personal information (e.g. saved albums, tracks, favorite artists, etc.) to provide a richer user experience.

closed 2016-09-28 by volumio