volumio/Volumio2 #647: Naming releases

kaffeeringe, 2016-08-15:
I really really like volumio. It's great project with great ideas and it is one of the few open-source project with a nice UI. I have the highest respect for your work, so please don't misunderstand my critizism:

"Release Candidate" means that the release is ready. Volumio2 is great, but it's far from ready.

I learned that if you are working on the software you call releases "alpha". 
You decide upon a roadmap and when all the features are in place that you wanted for your release, you proclaim a "feature freeze". No new features from this point on until the release. Only bug fixing. Now you call releases "beta". When all reported bugs are closed (or postponed) you release a "release candidate". So if no new major bugs are being reported THIS will be the final release. You release it again. Or you release the next release candidate.

I installed Volumio 2 because you called it a release candidate and I thought it had maybe some minor problems on exotic hardware. But I had trouble with almost every aspect of the software: Setting up the Wifi, disconnecting it from the LAN. The OTA-Update broke the Wifi and the NFS-mounting. I still haven't found out, why my NFS-share doesn't mount correctly (I can't rule out, that it's a server problem). Sound output is still set to audi jack, when I switched to HifiBerry. And so on. 

All that is okay for a beta release! But this is not a release candidate. 

I hope you see, that this not a critique of your work - only of your naming- and release policy ;-)

And I will try my best to help you!

WalterBorgstein, 2016-08-15:
👍🏻

volumio, 2016-08-15:
You're absolutely right. I've done pretty much everything in a rush, in order to get it out quickly... THe idea was to have a final set to release out, and then collect for bug reports (trust me, there are really looots of different combinations of hardware\usage\dacs etc...  So a big deal is having all those feedbacks... That being said, how would you suggest we could improve our naming scheme?

kaffeeringe, 2016-08-15:
Alpha = Show off first pieces of working code
Beta = No new features from this point. Only bug-fixing.
Release Candidate = No critical bugs. If nobody reports any critical bugs for a week or so, release it again as final version.

Make a roadmap of features, you want to include in every release. So ideas don't get lost and you don't get lost in too many ideas. 

Maybe you could strip down some of the features (don't show them in the UI) and focus on a basic set of features for 2.0.

yobenny, 2016-12-06:
Hello Guys, 

Congrats for all the job made. I'm found of HIFI and I enjoy volumio....except the laste release.

Just this message to encourage you to listen the release 2 on a HIFI high level system. The sound is so bad and you can hear some craks (I use NAS syno then Raspberry 2 (USB output) then my Hifi NAS)

I ve to get back to the 1.55  release. It s  petty because the V2 is much more smart, web radio are playing better.
Sorry for my bad news, I m sure that you ll improve the product in a goog way again and again
YO
PS : Also issue to mount NAS when the folder is on the PC / V2 doesn't find all the devices on the network

closed 2017-06-02 by volumio