volumio/Volumio2 #651: Albumart UI bug with Multiroom

skikirkwood, 2016-08-16:
The Multiroom Devices UI section is putting my Pi's IP address in front of the URL for Spotify albumart on the Playback screen:

[image]
[image]

balbuze, 2016-08-21:
yes I've got the same. And other thing, from a device I can't access an other one, I click on the screen is refreshed, but it stay on the same device. On the third device it's ok...
[image]

volumio, 2016-08-21:
This is a UI bug, could you please move it there?

closed 2016-08-21 by volumio