volumio/Volumio2 #654: Issues with Alarm Clock and with 'Add to playlist'

kiwiASRALL, 2016-08-17:
Hi all,

I reported an issue with function 'Alarm' and with 'Playlist'.

**Alarm:**
Add alarm, for example '08:00' and playlist 'test'.  All hours, music is playing (at [...] 07:00, 08:00, 09:00 [...]).  Only solution is delete alarm and not to use it.

**Playlist:**
Move to your local library, for me:
_Music Libray -> NAS -> Music_ 
On the menu button (who you can choose, Play/Add to queue/...) of an directory, choose "Add to playlist' and select a playlist, for me 'test'. 
Go to the playlist, it is empty. The solution is to select 'Add to queue' and in the queue 'Save queue as a playlist'. 

I use the latest release (0.928) on a RPI3. 

Thank you

nosferatujr, 2016-08-21:
Just to say that still on the new version released today. 0.979

fanciulli, 2016-09-13:
Working on it

fanciulli, 2016-09-13:
Patch missed fixing Add to queue

commit 279154d2 (2016-09-13): Fixes #654
commit 4d180484 (2016-09-13): Merge pull request #696 from volumio/issue_654

Fixes #654
commit 02b25441 (2016-09-13): Fixes #654
commit d0ccfd7e (2016-09-13): Merge pull request #697 from volumio/issue_654

Fixes #654
closed 2016-09-13 by volumio