volumio/Volumio2 #656: User interface labels are shown in stead of button text

WalterBorgstein, 2016-08-20:
[image]

Running on RPi3.
version: 0.978

possible root cause:
adding the NFS share to synology.
On a second try (fresh image, changed output to DAC, added NFS share) it did not happen :( So maybe unrelated.

volumio, 2016-08-20:
Please try to reload the page. Also, did you change the default language?

WalterBorgstein, 2016-08-20:
I did try reload, also did factory reset. No Luck.
Did not touch the language setting. I have had this happen also on the previous version. I assumed it was related to adding a NFS share (which is no working for me :( ), but I didn't manage to recreate the bug. Will post updates when it happens again.

I seem unable to find substantial logging on the image. the /var/log dir is rather empty. So I am unable to provide more details. Happy to help or provide evidence in any way.

volumio, 2016-08-20:
Which browser do you use? If you use chrome, please go to the inspector and then select console, see if you have some error there. 
The logs are stored in the journal. To see what goes on do:

sudo journalctl -f

balbuze, 2016-08-20:
It seems that this issue is due to a problem in /data/configuration/miscellanea/appearance/config.json
If you remove this file, it rework. But I agree it doesn’t tell why it occurs.

WalterBorgstein, 2016-08-20:
@volumio currently I'm using safari. Will give Chrome a run if the error re occurs.

WalterBorgstein, 2016-08-20:
@volumio The error game back after I removed a background using the GUI.
The following error is shown in the console.

[image]

WalterBorgstein, 2016-08-20:
I found some error in the journal files.

```

> Aug 20 14:17:41 volumio volumio[595]: info: CoreCommandRouter::executeOnPlugin: appearance , getBackgrounds
Aug 20 14:17:41 volumio volumio[595]: info: CoreCommandRouter::executeOnPlugin: appearance , getUiSettings
Aug 20 14:17:47 volumio volumio[595]: info: CoreCommandRouter::executeOnPlugin: appearance , deleteBackgrounds
Aug 20 14:17:47 volumio volumio[595]: /volumio/app/plugins/miscellanea/appearance/index.js:359
Aug 20 14:17:47 volumio volumio[595]: self.commandRouter.pushToastMessage('success', self.getI18NString('appearance_title'),
Aug 20 14:17:47 volumio volumio[595]: ^
Aug 20 14:17:47 volumio volumio[595]: TypeError: self.getI18NString is not a function
Aug 20 14:17:47 volumio volumio[595]: at /volumio/app/plugins/miscellanea/appearance/index.js:359:77
Aug 20 14:17:47 volumio volumio[595]: at FSReqWrap.oncomplete (fs.js:123:15)
```
```

WalterBorgstein, 2016-08-20:
The tip from @balbuze is working. After getting the label error, renaming the  /data/configuration/miscellanea/appearance/config.json file seems to work.

I made a differences overview so someone can determine what is misinterpreted. 

[image]

volumio, 2016-08-20:
Thanks, that was a bug, fixed here:
https://github.com/volumio/Volumio2/commit/913bfc0d191a11b9aef76c6ae8138df3bf6c4a87

I don't know if that related to the other one..

volumio, 2016-08-20:
what was the one working? the first or second?

WalterBorgstein, 2016-08-20:
_old was the failing one.

volumio, 2016-08-20:
Maybe I found it, changing to color breaks the thing

WalterBorgstein, 2016-08-20:
When selecting the dark background color it seems to break the config file.

WalterBorgstein, 2016-08-20:
Great minds think alike

volumio, 2016-08-20:
we're in the right direction. great!

volumio, 2016-08-20:
wahooo you've spotted and solved two bugs in a row... great work!

WalterBorgstein, 2016-08-20:
Thanks. Not sure if i fixed anything, but I like contributing by testing!

volumio, 2016-08-20:
Well it was a three men works... You spot it, balbuze suggests it, I do the dirty work
https://github.com/volumio/Volumio2-UI/commit/205fedce92852b1556aca6c310d9a3273cda2dee

volumio, 2016-08-20:
Can you try now to see if everything works as supposed?

WalterBorgstein, 2016-08-20:
How? ;)

volumio, 2016-08-20:
rm -rf /volumio/http/www/
git clone --depth 1 -b dist --single-branch https://github.com/volumio/Volumio2-UI.git build/$BUILD/root/volumio/http/www

WalterBorgstein, 2016-08-20:
Ok will give it a try ;). Need to reboot. I'm in a error state cycle it seems. I think these commands will fix that as well.

WalterBorgstein, 2016-08-20:
is it possible to overwrite the entire Volumio install this way? I crippled it.

volumio, 2016-08-20:
Nothing works at all? If it does, you can do a factory reset... If not it will take longer to fix it than to reflash it..

WalterBorgstein, 2016-08-20:
I think something broke in the Spotify plugin. I kept seeing Volumio restarting in the Journal. I could ssh in to the RP3, but the web interface would not come back. I'm lacking the in depth skills of Volumio to fix or reset things. So I did a refresh ;). Will plug it in next and kick off your commands.

volumio, 2016-08-20:
great , you probably changed language. Which is the next bug I'm going to tackle...

WalterBorgstein, 2016-08-20:
Not that I know ;)

WalterBorgstein, 2016-08-20:
1) Changing background colours (all) is working.
2) changing background image (all) is working

volumio, 2016-08-20:
Point n3 was fixed in the BE, which you don't have. 
Thanks, you helped me fix something I didn't have a clue about.  I'll prepare a new image now, and if that's ok release  it tomorrow

WalterBorgstein, 2016-08-20:
Great! I will continue to harass the interface until it brakes and than provide an issue. For now this day was lot's of fun! But I'm getting hungry. Talk to you later! Thanks for all your effort in creating and updating Volumio! Really like it.

volumio, 2016-08-20:
I released it as an update...

closed 2016-08-20 by volumio