volumio/Volumio2 #657: [WARN] Login failed: Cannot connect to Spotify/ [ERR ] Could not get the playlist container.

WalterBorgstein, 2016-08-20:
Added the Spotify Plugin.
Tested on RPi3, fresh install.

I use Facebook to logon to Spotify. So I had to create a password for devices. 
I entered the information in the Spotify plugin after installing (installing went perfect)

I can see general playlists, but can't seem to play anything.

> Aug 20 13:16:57 volumio volumio[10501]: 2016-08-20 13:16:57 [SPTF] 13:16:57.472 E [ap:3953] Connection error:  117
> Aug 20 13:16:57 volumio volumio[10501]: 2016-08-20 13:16:57 [SPTF] 13:16:57.472 I [ap:1752] Connecting to AP ap.spotify.com:443
> Aug 20 13:16:57 volumio volumio[595]: info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
> Aug 20 13:16:57 volumio volumio[595]: info: [1471699017479] CoreMusicLibrary::Adding element Spotify
> Aug 20 13:16:57 volumio volumio[595]: info: FIRST BRANCH
> Aug 20 13:17:02 volumio volumio[10501]: 2016-08-20 13:17:02 [SPTF] 13:17:02.476 E [ap:1694] AP Socket Error: Timeout reached (20000)
> Aug 20 13:17:02 volumio volumio[10501]: 2016-08-20 13:17:02 [SPTF] 13:17:02.477 E [ap:3953] Connection error:  117
> Aug 20 13:17:02 volumio volumio[10501]: 2016-08-20 13:17:02 [SPTF] 13:17:02.477 I [ap:1752] Connecting to AP ap.spotify.com:80
> Aug 20 13:17:03 volumio volumio[595]: info: Spotify access token expires at 1471702617468
> Aug 20 13:17:03 volumio volumio[595]: info: Spotify access token is *token removed*
> Aug 20 13:17:03 volumio volumio[595]: info: Spotify credentials grant success
> Aug 20 13:17:07 volumio volumio[10501]: 2016-08-20 13:17:07 [SPTF] 13:17:07.480 E [ap:1694] AP Socket Error: Timeout reached (20000)
> Aug 20 13:17:07 volumio volumio[10501]: 2016-08-20 13:17:07 [SPTF] 13:17:07.481 E [ap:3953] Connection error:  117
> Aug 20 13:17:07 volumio volumio[10501]: 2016-08-20 13:17:07 [SPTF] 13:17:07.481 I [ap:1752] Connecting to AP ap.spotify.com:4070
> Aug 20 13:17:12 volumio volumio[10501]: 2016-08-20 13:17:12 [SPTF] 13:17:12.485 E [ap:1694] AP Socket Error: Timeout reached (20000)
> Aug 20 13:17:12 volumio volumio[10501]: 2016-08-20 13:17:12 [SPTF] 13:17:12.485 E [ap:3953] Connection error:  117
> Aug 20 13:17:12 volumio volumio[10501]: 2016-08-20 13:17:12 [SPTF] 13:17:12.485 I [ap:1752] Connecting to AP ap.spotify.com:443
> Aug 20 13:17:17 volumio volumio[10501]: 2016-08-20 13:17:17 [SPTF] 13:17:17.488 E [ap:1694] AP Socket Error: Timeout reached (20000)
> Aug 20 13:17:17 volumio volumio[10501]: 2016-08-20 13:17:17 [SPTF] 13:17:17.488 E [ap:3953] Connection error:  117
> Aug 20 13:17:17 volumio volumio[10501]: 2016-08-20 13:17:17 [SPTF] 13:17:17.488 I [ap:1752] Connecting to AP ap.spotify.com:80
> Aug 20 13:17:22 volumio volumio[10501]: 2016-08-20 13:17:22 [SPTF] 13:17:22.491 E [ap:1694] AP Socket Error: Timeout reached (20000)
> Aug 20 13:17:22 volumio volumio[10501]: 2016-08-20 13:17:22 [SPTF] 13:17:22.491 E [ap:3953] Connection error:  117
> Aug 20 13:17:22 volumio volumio[10501]: 2016-08-20 13:17:22 [WARN] Login failed: Cannot connect to Spotify
> Aug 20 13:17:22 volumio volumio[10501]: 2016-08-20 13:17:22 [ERR ] Could not get the playlist container.
> Aug 20 13:17:22 volumio systemd[1]: spop.service: main process exited, code=killed, status=5/TRAP
> Aug 20 13:17:22 volumio systemd[1]: Unit spop.service entered failed state.

WalterBorgstein, 2016-08-20:
On playing from a "general" playlist, the following appeared in the journal log.

```
Aug 20 13:25:21 volumio volumio[595]: info: {"uri":"spotify:track:23L5CiUhw2jV1OIMwthR3S","service":"spop","name":"In the Name of Love","artist":"Martin Garrix","album":"In the Name of Love","type":"track","duration":195,"tracknumber":1,"albumart":"https://i.scdn.co/image/a1a2e08be66df92c75e46ad8330c9d4937bff209","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"}
Aug 20 13:25:21 volumio volumio[595]: info: [1471699521624] ControllerSpop::sendSpopCommand
Aug 20 13:25:21 volumio volumio[595]: info: ADDING DEFER FOR COMMAND play
Aug 20 13:25:21 volumio volumio[595]: COMMON.PLAY_TITLE
Aug 20 13:25:21 volumio volumio[595]: [ 'COMMON', 'PLAY_TITLE' ]
Aug 20 13:25:21 volumio volumio[595]: COMMON.PLAY_TEXT
Aug 20 13:25:21 volumio volumio[595]: [ 'COMMON', 'PLAY_TEXT' ]
Aug 20 13:25:21 volumio volumio[595]: info: SPOP command error:
Aug 20 13:25:21 volumio volumio[595]: info:  Error: This socket has been ended by the other party
Aug 20 13:25:21 volumio volumio[595]: at Socket.writeAfterFIN [as write] (net.js:285:12)
Aug 20 13:25:21 volumio volumio[595]: at applyWithContext (/data/plugins/music_service/spop/node_modules/fast.js/function/applyWithContext.js:15:22)
Aug 20 13:25:21 volumio volumio[595]: at /data/plugins/music_service/spop/node_modules/fast.js/function/bind.js:63:14
Aug 20 13:25:21 volumio volumio[595]: at onBoundPromise (/data/plugins/music_service/spop/node_modules/kew/kew.js:834:10)
Aug 20 13:25:21 volumio volumio[595]: at ncall (/data/plugins/music_service/spop/node_modules/kew/kew.js:816:44)
Aug 20 13:25:21 volumio volumio[595]: at Object.nfcall (/data/plugins/music_service/spop/node_modules/kew/kew.js:803:16)
Aug 20 13:25:21 volumio volumio[595]: at Promise._successFn (/data/plugins/music_service/spop/index.js:1031:16)
Aug 20 13:25:21 volumio volumio[595]: at nextTickCallback (/data/plugins/music_service/spop/node_modules/kew/kew.js:47:28)
Aug 20 13:25:21 volumio volumio[595]: at _combinedTickCallback (internal/process/next_tick.js:67:7)
Aug 20 13:25:21 volumio volumio[595]: at process._tickCallback (internal/process/next_tick.js:98:9)
```

WalterBorgstein, 2016-08-20:
Fresh install, install of Spotify plugin:
- I can see the public playlist, not private.

Error appears in the log:

>  info:  Error: This socket has been ended by the other party

Following is the log of clicking a random song:

```
Aug 20 19:06:48 volumio volumio[545]: info: CoreCommandRouter::volumioAddQueueItems
Aug 20 19:06:48 volumio volumio[545]: info: [1471720008072] CoreStateMachine::addQueueItems
Aug 20 19:06:48 volumio volumio[545]: info: [1471720008073] CorePlayQueue::addQueueItems
Aug 20 19:06:48 volumio volumio[545]: info:  uri=spotify:track:3hfN1wQSiLPa6b1RX5Kfbj, title=WHISTLE, service=spop
Aug 20 19:06:48 volumio volumio[545]: info: First index is 0
Aug 20 19:06:48 volumio volumio[545]: info: ADDING THIS ITEM TO QUEUE: {"uri":"spotify:track:3hfN1wQSiLPa6b1RX5Kfbj","title":"WHISTLE","service":"stop"}
Aug 20 19:06:48 volumio volumio[545]: info: Exploding uri spotify:track:3hfN1wQSiLPa6b1RX5Kfbj in service spop
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013841] CorePlayQueue::saveQueue
Aug 20 19:06:53 volumio volumio[545]: info: Adding item to queue: [{"uri":"spotify:track:3hfN1wQSiLPa6b1RX5Kfbj","service":"spop","name":"WHISTLE","artist":"Blackpink","album":"SQUARE ONE","type":"track","duration":211,"tracknumber":1,"albumart":"https://i.scdn.co/image/119a66360da4b3f446d709b19b4a91b319daaebc","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"}]
Aug 20 19:06:53 volumio volumio[545]: info: CoreCommandRouter::volumioPushQueue
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013844] InterfaceWebUI::pushQueue
Aug 20 19:06:53 volumio volumio[545]: info: [{"uri":"spotify:track:3hfN1wQSiLPa6b1RX5Kfbj","service":"spop","name":"WHISTLE","artist":"Blackpink","album":"SQUARE ONE","type":"track","duration":211,"tracknumber":1,"albumart":"https://i.scdn.co/image/119a66360da4b3f446d709b19b4a91b319daaebc","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"}]
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013846] CoreStateMachine::updateTrackBlock
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013847] CorePlayQueue::getTrackBlock
Aug 20 19:06:53 volumio volumio[545]: info: CoreCommandRouter::volumioPlay
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013848] CoreStateMachine::play
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013848] CoreStateMachine::setConsumeUpdateService undefined
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013849] CoreStateMachine::stop
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013849] CoreStateMachine::setConsumeUpdateService undefined
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013850] CoreStateMachine::play
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013851] CoreStateMachine::setConsumeUpdateService undefined
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013851] CorePlayQueue::getTrack 0
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013852] CoreStateMachine::startPlaybackTimer
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013853] ControllerSpop::clearAddPlayTrack
Aug 20 19:06:53 volumio volumio[545]: info: {"uri":"spotify:track:3hfN1wQSiLPa6b1RX5Kfbj","service":"spop","name":"WHISTLE","artist":"Blackpink","album":"SQUARE ONE","type":"track","duration":211,"tracknumber":1,"albumart":"https://i.scdn.co/image/119a66360da4b3f446d709b19b4a91b319daaebc","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"}
Aug 20 19:06:53 volumio volumio[545]: info: [1471720013855] ControllerSpop::sendSpopCommand
Aug 20 19:06:53 volumio volumio[545]: info: ADDING DEFER FOR COMMAND play
Aug 20 19:06:53 volumio volumio[545]: COMMON.PLAY_TITLE
Aug 20 19:06:53 volumio volumio[545]: [ 'COMMON', 'PLAY_TITLE' ]
Aug 20 19:06:53 volumio volumio[545]: COMMON.PLAY_TEXT
Aug 20 19:06:53 volumio volumio[545]: [ 'COMMON', 'PLAY_TEXT' ]
Aug 20 19:06:53 volumio volumio[545]: info: SPOP command error:
Aug 20 19:06:53 volumio volumio[545]: info:  Error: This socket has been ended by the other party
Aug 20 19:06:53 volumio volumio[545]: at Socket.writeAfterFIN [as write] (net.js:285:12)
Aug 20 19:06:53 volumio volumio[545]: at applyWithContext (/data/plugins/music_service/spop/node_modules/fast.js/function/applyWithContext.js:15:22)
Aug 20 19:06:53 volumio volumio[545]: at /data/plugins/music_service/spop/node_modules/fast.js/function/bind.js:63:14
Aug 20 19:06:53 volumio volumio[545]: at onBoundPromise (/data/plugins/music_service/spop/node_modules/kew/kew.js:834:10)
Aug 20 19:06:53 volumio volumio[545]: at ncall (/data/plugins/music_service/spop/node_modules/kew/kew.js:816:44)
Aug 20 19:06:53 volumio volumio[545]: at Object.nfcall (/data/plugins/music_service/spop/node_modules/kew/kew.js:803:16)
Aug 20 19:06:53 volumio volumio[545]: at Promise._successFn (/data/plugins/music_service/spop/index.js:1031:16)
Aug 20 19:06:53 volumio volumio[545]: at nextTickCallback (/data/plugins/music_service/spop/node_modules/kew/kew.js:47:28)
Aug 20 19:06:53 volumio volumio[545]: at _combinedTickCallback (internal/process/next_tick.js:67:7)
Aug 20 19:06:53 volumio volumio[545]: at process._tickCallback (internal/process/next_tick.js:98:9)

Aug 20 19:06:48 volumio volumio[545]: info: CoreCommandRouter::volumioAddQueueItems
Aug 20 19:06:48 volumio volumio[545]: info: [1471720008072] CoreStateMachine::addQueueItems
Aug 20 19:06:48 volumio volumio[545]: info: [1471720008073] CorePlayQueue::addQueueItems
Aug 20 19:06:48 volumio volumio[545]: info:  uri=spotify:track:3hfN1wQSiLPa6b1RX5Kfbj, title=WHISTLE, service=spop
[... cut ...]
Aug 20 19:06:53 volumio volumio[545]: at Object.nfcall (/data/plugins/music_service/spop/node_modules/kew/kew.js:803:16)
Aug 20 19:06:53 volumio volumio[545]: at Promise._successFn (/data/plugins/music_service/spop/index.js:1031:16)
Aug 20 19:06:53 volumio volumio[545]: at nextTickCallback (/data/plugins/music_service/spop/node_modules/kew/kew.js:47:28)
Aug 20 19:06:53 volumio volumio[545]: at _combinedTickCallback (internal/process/next_tick.js:67:7)
Aug 20 19:06:53 volumio volumio[545]: at process._tickCallback (internal/process/next_tick.js:98:9)

```

balbuze, 2016-08-20:
are you using Spotify with the same account from an other device ? (I mean in the same time)

WalterBorgstein, 2016-08-20:
No. I do have a family account, that has 5 users for the account. Maybe one of them is using it. I can't be sure.

balbuze, 2016-08-20:
ok but I mean log with the same credential. I 've got a family account too, but each member have its own credential. In the case a account is used, I don't think you're able to use it in the same time from a other device. But is just a question as I never encounter such problem. But I don't log with a facebook account...

WalterBorgstein, 2016-08-20:
No for sure I'm not using my credentials somewhere else (like iPhone or KODI).
So it not a general issue then. That makes me worry about my install :(.

WalterBorgstein, 2016-08-20:
volumio@volumio:~$ nslookup ap.spotify.com               
Server:		8.8.4.4
Address:	8.8.4.4#53

Non-authoritative answer:
ap.spotify.com	canonical name = ap.gslb.spotify.com.
Name:	ap.gslb.spotify.com
Address: 193.182.7.82
Name:	ap.gslb.spotify.com
Address: 194.132.162.151
Name:	ap.gslb.spotify.com
Address: 194.14.177.22
Name:	ap.gslb.spotify.com
Address: 194.132.162.149
Name:	ap.gslb.spotify.com
Address: 194.14.177.37
Name:	ap.gslb.spotify.com
Address: 194.132.204.162
Name:	ap.gslb.spotify.com
Address: 194.132.162.136
Name:	ap.gslb.spotify.com
Address: 194.14.177.18

I can nslookup the Spotify adres. (I performed a `sudo apt-get install dnsutils` to get NSLOOKUP for that.

skikirkwood, 2016-08-20:
From the log files it looks to me like there's some problem connecting to the Spotify spop daemon on your machine.

Look at your process list (ps -ef) and you should have the spopd process running:

/usr/bin/spopd -f -c /etc/spopd.conf

If not, try starting it and restarting Volumio's node.js process (killall node).  If spopd is running try killing it, restarting Volumio and see if that helps.  

The Spotify plugin tries to connect to the spop process on port 6602, and for some reason that is not happening on your machine.

WalterBorgstein, 2016-08-20:
Thanks.

I could not find the `/usr/bin/spopd -f -c /etc/spopd.conf` so I executed `killall node`
After that, Volumio restarts (i hear the startup sound)
I can see the `/usr/bin/spopd -f -c /etc/spopd.conf`
couple seconds later using the ps -ef, it is gone.

Following appears in the journal log.
```
Aug 20 21:28:36 volumio systemd[1]: spop.service: main process exited, code=killed, status=5/TRAP
Aug 20 21:28:36 volumio systemd[1]: Unit spop.service entered failed state.
```

skikirkwood, 2016-08-20:
Ok, it's good we definitely narrowed down the issue to spopd.  I worked on the recent enhancements to the Spotify Plugin that uses the new Spotify Web API, but not that familiar with spopd.  I did find the error message you got listed as an open issue with spop however:

https://github.com/Schnouki/spop/issues/22

Can you delete your queue to make sure nothing is starting up before you try using Spotify.  I'm afraid I don't think I can help diagnose the problem beyond this.

WalterBorgstein, 2016-08-20:
To make sure I didn't miss any log entries. I did the killall node again.
From that point (including me clearing the queue as requested) is pasted below.

```
Aug 20 21:43:02 volumio volumio[546]: [1471729382407] ---------------------------- Client requests Volumio Clear Queue
Aug 20 21:43:02 volumio volumio[546]: info: CoreCommandRouter::volumioClearQueue
Aug 20 21:43:02 volumio volumio[546]: info: [1471729382409] CoreStateMachine::ClearQueue
Aug 20 21:43:02 volumio volumio[546]: info: [1471729382410] CoreStateMachine::stop
Aug 20 21:43:02 volumio volumio[546]: info: [1471729382411] CoreStateMachine::setConsumeUpdateService undefined
Aug 20 21:43:02 volumio volumio[546]: info: [1471729382413] CorePlayQueue::clearPlayQueue
Aug 20 21:43:02 volumio volumio[546]: info: [1471729382414] CorePlayQueue::saveQueue
Aug 20 21:43:02 volumio volumio[546]: info: CoreCommandRouter::volumioPushQueue
Aug 20 21:43:02 volumio volumio[546]: info: [1471729382418] InterfaceWebUI::pushQueue
Aug 20 21:43:02 volumio volumio[546]: info: []
Aug 20 21:43:02 volumio volumio[546]: info: [1471729382422] ------------------------------ 15ms
Aug 20 21:43:42 volumio sudo[2134]: volumio : TTY=pts/1 ; PWD=/home/volumio ; USER=root ; COMMAND=/usr/bin/killall node
Aug 20 21:43:42 volumio sudo[2134]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
Aug 20 21:43:42 volumio sudo[2134]: pam_unix(sudo:session): session closed for user root
Aug 20 21:43:42 volumio mpd[578]: client: [0] closed
Aug 20 21:43:42 volumio systemd[1]: volumio.service holdoff time over, scheduling restart.
Aug 20 21:43:42 volumio systemd[1]: Stopping volumio.service...
Aug 20 21:43:42 volumio systemd[1]: Starting volumio.service...
Aug 20 21:43:42 volumio systemd[1]: Started volumio.service.
Aug 20 21:43:45 volumio volumio[2144]: info: -------------------------------------------
Aug 20 21:43:45 volumio volumio[2144]: info: -----            Volumio2              ----
Aug 20 21:43:45 volumio volumio[2144]: info: -------------------------------------------
Aug 20 21:43:45 volumio volumio[2144]: info: -----          System startup          ----
Aug 20 21:43:45 volumio volumio[2144]: info: -------------------------------------------
Aug 20 21:43:45 volumio volumio[2144]: info: Plugin folders cleanup
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning into folder /volumio/app/plugins/
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning category audio_interface
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning category miscellanea
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning category music_service
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning category plugins.json
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning category system_controller
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning category user_interface
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning into folder /data/plugins/
Aug 20 21:43:45 volumio volumio[2144]: info: Scanning category music_service
Aug 20 21:43:45 volumio volumio[2144]: info: Plugin folders cleanup completed
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugins from folder /volumio/app/plugins/
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugins from folder /data/plugins/
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugin "system"...
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugin "appearance"...
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugin "network"...
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugin "services"...
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugin "alsa_controller"...
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugin "volumio_command_line_client"...
Aug 20 21:43:45 volumio volumio[2144]: info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
Aug 20 21:43:45 volumio volumio[2144]: info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
Aug 20 21:43:45 volumio volumio[2144]: info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
Aug 20 21:43:45 volumio volumio[2144]: info: Loading plugin "volumiodiscovery"...
Aug 20 21:43:45 volumio volumio[2144]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Aug 20 21:43:45 volumio node[2144]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Aug 20 21:43:45 volumio node[2144]: *** WARNING *** Please fix your application to use the native API of Avahi!
Aug 20 21:43:45 volumio node[2144]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Aug 20 21:43:45 volumio node[2144]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
Aug 20 21:43:45 volumio node[2144]: *** WARNING *** Please fix your application to use the native API of Avahi!
Aug 20 21:43:45 volumio node[2144]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
Aug 20 21:43:45 volumio volumio[2144]: *** WARNING *** Please fix your application to use the native API of Avahi!
Aug 20 21:43:45 volumio volumio[2144]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
[... cut ...]
Aug 20 21:44:17 volumio volumio[2200]: 2016-08-20 21:44:17 [SPTF] 21:44:17.710 E [ap:3953] Connection error:  117
Aug 20 21:44:17 volumio volumio[2200]: 2016-08-20 21:44:17 [WARN] Login failed: Cannot connect to Spotify
Aug 20 21:44:17 volumio volumio[2200]: 2016-08-20 21:44:17 [ERR ] Could not get the playlist container.
Aug 20 21:44:17 volumio systemd[1]: spop.service: main process exited, code=killed, status=5/TRAP
Aug 20 21:44:17 volumio systemd[1]: Unit spop.service entered failed state.

```

skikirkwood, 2016-08-20:
Walter, you are getting a bunch of 117 connection errors, which I believe is coming from spopd trying to connect to Spotify.  People have had this before:

https://ubuntuforums.org/showthread.php?t=1035338

Have you ever successfully run Spotify on Volumio on this Rpi?

skikirkwood, 2016-08-20:
Here's a posting from Spotify support regarding what to check when a 117 error occurs:

https://community.spotify.com/t5/Desktop-Linux-Windows-Web-Player/Troubleshooting-connection-issues/td-p/104003

WalterBorgstein, 2016-08-20:
No I never got Spotify to work. Just started with RPB3 and Volumio. ;)

WalterBorgstein, 2016-08-20:
Spotify does work on my iPhone, and Windows 10 PC. They are all going thru the same router. So it should work connection wise.
(btw, the other accounts are not running as I am testing)

skikirkwood, 2016-08-20:
A final thing I would try checking, if you haven't, is running the Spotify Web player (play.spotify.com) on your PC and ensure that works, in case you are using the Spotify Windows app.  I'm pretty sure it's some kind of networking issue blocking the connection, but I'm out of ideas as to how to diagnose it further, beyond perhaps logging out of your Spotify account on the Web, and logging back in, which might reset something on Spotify's servers.  Final thing is I would change your Spotify password.

Here's one more odd posting on the Spotify forums:

joaquimabc
joaquimabc Concert Regular  ‎2015-10-11 02:46 PM
I have a family account. Since i cancel and adn sing to add a new user i get a error 117 message in linux... In android the downlowd to play off-line stay in waiting...
 
My firewall is off, and, other users play normaly....
 
I think this is a change in the server...
 
Is this?

WalterBorgstein, 2016-08-20:
Thanks for all your help! I will try this. 
- I reset my router
- created a new account for Spotify
- Gave that new account a family invite and am now using that one
Still nothing.

shawaj, 2016-08-22:
I had the same issue when trying to log in to Spotify with email address instead of username.

I never created the username so didn't know it existed but accidentally happened upon it in Spotify settings and tried it and it worked. Could this have been your issue?

WalterBorgstein, 2016-08-22:
No, I tried all combinations. User number, user account, email etc. I'm currently logged in as my Spotify number.

closed 2017-06-02 by volumio

renavspainatal, 2019-11-15:
For the Facebook users coming here for an answer, you need a **device password** and a **username**.  **Not** your facebook login/password!
Go to https://www.spotify.com/au/account/set-device-password/ for your username (sequence of numbers) and to set a device password.