volumio/Volumio2 #659: NAS path difference between CIFS and NFS

balbuze, 2016-08-20:
I made some tests and mount my NAS using CIFS and NFS
It appears that for CIFS mount the PATH musn't start with / but for NFS it must start with /
It's a bit confusing.
[image]

WalterBorgstein, 2016-08-20:
I'm using NFS and the path doesn't need the /.
What happens without the /?

[image]

balbuze, 2016-08-20:
Humm... maybe it is to the way my NAS shares folder... If I omit the / it does nothing, and appears as not mounted.

I see you still have button bugs...

WalterBorgstein, 2016-08-20:
Yeah they came back. Added a screenshot of Chrome Console for dev skilled people. ;)

WalterBorgstein, 2016-08-20:
What NAS do you have? Maybe you can add some to the documentation draft I made concerning Synology Shares [https://github.com/volumio/Volumio2/issues/641#issuecomment-241202422](url)

balbuze, 2016-08-20:
have you noticed a particular action for that ? Have you change something in appareance ? or nothing noticable ?

balbuze, 2016-08-20:
my nas is a dlink share center dns 320

closed 2016-09-28 by volumio