volumio/Volumio2 #668: Wifi connecting and then disconnecting immediately after

hfatgit, 2016-08-21:
Wifi does connect when booting the system. It stays connected for a short period of time, retrieves the DHCP address, then disconnects from AP.  

Volumio 2 not usable at all using Wifi

WalterBorgstein, 2016-08-21:
Thats a harsh statement. I have it working now on wifi. No issues. Kept it running all night long to test stability. Could you get some logs from the Volumio journal? Maybe some devs can spot weird activity. 

I gave the volumio a fixed IP in the dhcp. Maybe that helps?

hfatgit, 2016-08-21:
Just stating the facts, nothing harsh here.  Wlan0 keeps the DHCP IP but cannot access the network. 
Static IP shows the same behaviour. 

Works like a charm when connected with eth0.
 
Version 1.5 did the Wifi job without problems.

WalterBorgstein, 2016-08-21:
Why close the issue?

hfatgit, 2016-08-21:
oops. Wurstfinger on mobile device

WalterBorgstein, 2016-08-21:
;)

volumio, 2016-08-21:
There is a setting in wireless.js that triggers hotspot mode if signal quality is not sufficient to get good performances.  At this point it seems that its too aggressive

hfatgit, 2016-08-21:
Qualtiy 96/100 Signal 94/100. Too low?

volumio, 2016-08-21:
Could you post the output of sudo journalctl - f?

hfatgit, 2016-08-21:
My current settings for the wifi are accoring to issue 164 like
 /etc/wpa_supplicant/wpa_supplicant_mywlan.conf:
network={
    ssid="MYSSID"
    psk="passphrase"
}

/etc/network/interfaces:
auto wlan0
iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant/wpa_supplicant_mywlan.conf

I tripplechecked the psk: phrase is correct 

[journal.txt](https://github.com/volumio/Volumio2/files/428873/journal.txt)
Not much contents and the last few lines repeating over and over

hfatgit, 2016-08-21:
Rebooted the Pi and logging the journalctl -f output for a while. Will attach the outcome soon.

hfatgit, 2016-08-21:
Started journalctl-f > journal.txt immediately after login was possible

[journal.txt](https://github.com/volumio/Volumio2/files/428888/journal.txt)

Callenberg, 2016-08-21:
I can confirm problems with the Wi-Fi connection, it is even worse with 0.979 than with 0.976.
I got it working by (after the connecting procedure):
1. turn off and turn on the Raspberry Pi
2. turn off
3. disconnect Ethernet cable
4. turn on

You can not just remove the Ethernet cable any more (as you could do using Volumio 1.55).
There were less turn ons and turn offs in the 0.976 version. 
You also have to refresh the pages manually, otherwise nothing seems to happen. The hotspot box disappears at an early stage in the procedure as well.

closed 2017-06-02 by volumio

christemporale, 2017-09-09:
I am running the latest version of Volumio: 2.246. I configured ethernet and wlan, both with static IP.
As long as an Ethernet cable is attached, I can access the GUI from both IPs and all works fine. Without ethernet cable, the system looses the wlan connection. After the boot, the wifi connection is lost almost immediately. I don't have this issue with runeadio