volumio/Volumio2 #676: Output device resets to Audio Jack after restart [Playback Options]

Callenberg, 2016-08-24:
In the Audio Output configuration the Output Device is reset to Audio Jack after restart.

This error does not occur every time the Volumio is restarted, which is unfortunate since it makes things harder. After some testing I have found out that it is more likely to occur if Volumio is in play mode when the Volumio is restarted.

I have restarted the Volumio by using both the Restart option and the Power Off option. Both ways may cause the error to occur. Turning off the Volumio by disconnecting the power makes no difference, the error still occurs.

System version: 0.979, Raspberry Pi 3, IQaudio Pi-DigiAMP+, 19V 3.42A, Class 6 8GB SD card
Audio Output settings entered by me: [IQAudIO DAC  /I2S DAC ON / IQaudio Pi-DigiAMP+]

volumio, 2016-08-24:
It's the same as this https://github.com/volumio/Volumio2/issues/615

Could you apply the suggested fix and see if that  works?

Callenberg, 2016-08-24:
Thanks for the quick response, and it is already fixed. Great!

Sorry that I didn't read #615 first.

closed 2016-08-24 by Callenberg