volumio/Volumio2 #677: Update from 0.976 to 0.979

kiwiASRALL, 2016-08-25:
Hi,

I have update volumio from the webui.

After reboot, the webui doesn't work.

After reading lot of forum posts, I have found that iptables redirect port 3000 to 80. Then:

```
root@volumio:~# iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
```

No documentation on your website/wiki... Google search and I've found:
https://github.com/volumio/Build/blob/master/volumio/etc/iptables.rules

I've applied same rules in the same file and restored rules with iptables. Works!

I hope it could help someone. Maybe this is a bug from the upgrade ? Or just me ?

And excuse my english...

kpmbdr, 2016-08-29:
my i know what you iptable contain?

kiwiASRALL, 2016-08-29:
This: https://github.com/volumio/Build/blob/master/volumio/etc/iptables.rules

```
volumio@volumio:~$ cat /etc/iptables.rules 
*nat
:PREROUTING ACCEPT [25:4112]
:INPUT ACCEPT [74:6612]
:OUTPUT ACCEPT [66:20563]
:POSTROUTING ACCEPT [66:20563]
-A PREROUTING -p tcp -m tcp --dport 80 -j REDIRECT --to-ports 3000
COMMIT

```

volumio, 2016-08-31:
Are you connected via LAN or wireless?

kiwiASRALL, 2016-09-03:
I'm connected via LAN

volumio, 2016-09-03:
Is LAN connected before turning the PI on? 
Are you on DHCP or Static ? 
Did  you manually edit /etc/network/interfaces ?

kiwiASRALL, 2016-09-03:
LAN is connected before turning on the PI, yes.

It is static assignation from my router based on mac adress. No change directly on the Pi.

Nop.

This issue was arrived just after upgrade volumio from web ui.

volumio, 2016-09-03:
It's quite tricky to understand what did go wrong...
Could you please paste the output of : 

df -h 

Thank you, it's very useful for us...

kiwiASRALL, 2016-09-03:
```
volumio@volumio:~$ df -h
Sys. de fichiers Taille Utilisé Dispo Uti% Monté sur
/dev/mmcblk0p2     2,0G    813M  1,1G  43% /imgpart
/dev/loop0         263M    263M     0 100% /static
overlay             12G    1,3G   10G  12% /
devtmpfs           479M       0  479M   0% /dev
tmpfs              487M    4,0K  487M   1% /dev/shm
tmpfs              487M     25M  462M   6% /run
tmpfs              5,0M    4,0K  5,0M   1% /run/lock
tmpfs              487M       0  487M   0% /sys/fs/cgroup
tmpfs              487M     60K  487M   1% /tmp
tmpfs              487M       0  487M   0% /var/spool/cups
tmpfs              487M       0  487M   0% /var/cache/apt/archives
tmpfs              487M    3,6M  483M   1% /var/log
tmpfs              487M       0  487M   0% /var/spool/cups/tmp
/dev/mmcblk0p1      61M     29M   33M  47% /boot
//TEKYLA/Music     1,9T    1,4T  504G  73% /mnt/NAS/Music
tmpfs               98M       0   98M   0% /run/user/1000

```

No problem ;-)

futuralogic, 2016-09-08:
I also had an issue upgrading between these versions from the WebUI.
I use a wifi usb dongle (EDIMax?) and have had to set static IP (DHCP doesn't seem to work) on volumio 2.
After the upgrade, it can't get on the network. Having hooked it up to my television to get details it looks like immediately after boot the static IP is setup from /etc/network/interfaces. I confirmed via cmdline doing ifconfig.
Several seconds later ifconfig returns an IP address of 192.168.211.1 which is not my network. My subnet is 192.168.1.x
Not sure what happened. Planning to flash back to 976 and reload my custom /interfaces file in order to keep it working.
Rasp Pi Model B w/ Hifiberry DAC, usb dongle wifi.
Thanks.

Finetuned, 2016-09-08:
Same as  (@kiwiASRALL) happened to me today using thie following build: IQ_volumio-0.861-2016-03-25-pi... 

```
df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  2.0G  777M  1.2G  41% /imgpart
/dev/loop0      263M  263M     0 100% /static
overlay         5.1G   12M  4.8G   1% /
devtmpfs        479M     0  479M   0% /dev
tmpfs           487M     0  487M   0% /dev/shm
tmpfs           487M  4.7M  483M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           487M     0  487M   0% /sys/fs/cgroup
tmpfs           487M   12K  487M   1% /tmp
tmpfs           487M     0  487M   0% /var/spool/cups
tmpfs           487M     0  487M   0% /var/cache/apt/archives
tmpfs           487M   24K  487M   1% /var/log
tmpfs           487M     0  487M   0% /var/spool/cups/tmp
/dev/mmcblk0p1   61M   28M   34M  46% /boot
tmpfs            98M     0   98M   0% /run/user/1000
```
and 
```
cat /etc/iptables.rules 
# Generated by iptables-save v1.4.21 on Thu Sep  8 15:14:43 2016
*nat
:PREROUTING ACCEPT [397:49492]
:INPUT ACCEPT [402:48808]
:OUTPUT ACCEPT [116:35713]
:POSTROUTING ACCEPT [116:35713]
[8:512] -A PREROUTING -p tcp -m tcp --dport 80 -j REDIRECT --to-ports 3000
COMMIT
```

volumio, 2016-09-08:
IQ Volumio? Where did you get it? 
It's super old, don't bother using it!

Finetuned, 2016-09-09:
@volumio From the Pi-DigiAMP+ site: http://www.iqaudio.com/downloads/

futuralogic, 2016-09-11:
I finally had time to test this, so I re-imaged .979 which I downloaded from the volumio.org homepage.

I have a custom /etc/network/interfaces file to connect to my wifi hidden SSID.  With volumio2 I have found that I need to run static IP as DHCP does not function correctly.

With .979, it seems possibly the hotspot functionality is interfering and overrides the network and I end up with an IP address on wifi of 192.168.211.1 which is not routable to my 192.168.1.x subnet.

I would like to go back to .976 but do not have an image downloaded because I self-updated to .976 via the web interface.  Now I'm stuck downloading .928 from Github.  I'm presuming if I try to update from .928 it will take me to .979 which will break.

It would be preferred if the system did not jack with my network settings.  Is there any way to disable the hotspot functionality by default and if someone wants to use it, they enable, reboot and go along their way.  I don't need that functionality, and it appears to be conflicting.

So back to .928 I will go.

xipmix, 2016-09-14:
> Is there any way to have all updates downloadable somewhere? Presumably they are available because the web-updater can grab them? 

I tried to figure this out in #543 . There was a problem back then that the image was using the 'dev' branch out of git and that stopped getting updated... Now it seems to be tracking master better so 'cd /volumio; git pull' might work. Not sure if you have to rebuild ('npm install').

closed 2017-06-02 by volumio