volumio/Volumio2 #680: Mixer control dropdown is empty (Raspberry Pi + Hifiberry DAC)

Atrejoe, 2016-08-28:
I wanted to see if I could control the volume of playback by changing the mixer control. (I remember a hardware vs software choice in Volumio 1).

The dropdown where I'd expect this choice is empty, see:
[image]

(Volumio version 0.979 on Raspberry PI 1 B, Hifiberry DAC, old model B)

balbuze, 2016-08-28:
Volumio2 does not yet support software mixer. It's because it's empty in your screenshot. It 's going to be added.

Atrejoe, 2016-08-30:
Sounding like a user here, suggestions:
- Shouldn't the dropdown be disabled when there is no choice (single option)
- Shouldn't the dropdown be replaced by a label stating there is no mixer control available in my case

I don't see software volume control on the roadmap (https://github.com/volumio/Volumio2), any ETA?

Sjit, 2016-09-21:
I have the same problem. 
Volumio version 0.979 on Raspberry PI 1 B, Hifiberry DAC, old model B
For my current use, mediaplayer for the kids, this is a significant problem since i dont have a hardware remote volume control.

volumio, 2016-09-24:
Fixed

closed 2016-09-24 by volumio

Atrejoe, 2016-09-26:
Great to hear, thanks!