volumio/Volumio2 #682: GMusic in Architecture Overview

dumyan, 2016-08-30:
In the [Architecture Overview](https://github.com/volumio/Volumio2/wiki#architecture-overview) GMusic controller is mentioned (as possible implementation in the future). Is this a planned implementation for existing controller?

Also, what are the pros and cons between plugin and having a controller for a music service?

I'm interested in developing this feature, so tips, tricks and points are welcomed.

balbuze, 2016-08-30:
I you want to write a plugin, you're welcome. A doc to start (to be completed) is here : https://volumio.github.io/docs/Plugin_System/Plugin_System_Overview.html .
And a way is to examine already existing app and plugin. have a look here : https://github.com/volumio/volumio-plugins/tree/master/plugins/music_service and of course : https://github.com/volumio/Volumio2/tree/master/app/plugins

dumyan, 2016-08-31:
Thanks for the pointers.

volumio, 2016-08-31:
Also, a good module you could use..
https://www.npmjs.com/package/playmusic

Tihmann, 2016-09-14:
Also this might be helpfull
https://github.com/volumio/Volumio2/issues/555

vanita5, 2017-02-07:
Looks like no one started working on this yet? If so, I'll start writing a plugin tomorrow.

fanciulli, 2017-02-08:
@vanita5 as far as i know no one is writing the plugin. You're welcome.

vanita5, 2017-02-09:
Hello everyone, time for a small update on the progress :)

I started writing the plugin and came pretty far. For now you can view playlists and play a sample track. However, I came across some limits.
I use [GMusicProxy](https://github.com/diraimondo/gmusicproxy/) and I realized that it is kinda limited too late. For example, I can't get information of a track by its id which is necessary for a correct implementation.

I just now discovered this module: [https://github.com/diraimondo/gmusicproxy/](https://github.com/diraimondo/gmusicproxy/) which looks far more complete, but I have to rewrite core functionality of the plugin.

Now for some questions.

Is it possible to paginate in playlists e.g.? Does Volumio support this anyhow? Becaus I happen to have some big playlists (~1000 tracks) and it takes too long to load them all at once.

Also, is anyone interested in Last.fm scrobbling? I'm thinking about including this in the Google Music plugin or as a seperate plugin if that works out.

volumio, 2017-02-09:
Hi, unfortunately we don't have pagination support... 
What about integrating last.fm scrobbling directly to volumio itself? This should be pretty straightforwarding... Let me know if you want to cooperate on this! 

And thanks for the plugin I'll give it a try! 

mi

vanita5, 2017-02-09:
@volumio Sounds great. I'll start rewriting the Goolge Music plugin next week and after that I might look into Last.fm.

amalic, 2017-02-14:
hi vanita, just wanted to see where my feature request went ;)

haven't had time to do any stuff myself. AFAIK one of the MPD console frontends (either ncmpc or ncmpcpp) resolved the title infos by loading the playlist i generated with gmusicproxy

this might be a hint into the right direction. if it's possible in cpp it should be possible in nodejs too i guess.

closed 2017-06-02 by volumio