volumio/Volumio2 #684: List of Network Drives

tim-felix-lange, 2016-09-03:
After setting up a connection to my NAS, the drive does not show up under Settings->My Music-> Network Drives. However, it does mount under /mnt/NAS, it does scan the content and it does show up in the Library, it's just missing in the settings, e.g. making it impossible to remove it via the WebUI.

volumio, 2016-09-03:
Is it CIFS or NFS?

tim-felix-lange, 2016-09-03:
NFS

xipmix, 2016-10-16:
Tim, is this working for you now (version 2.000)? If not could you try the following
 * grab a clean volumio image
 * set up basic networking and reboot
 * set up network drive
 * save the logs and post them here in a .zip
```
$ sudo journalctl > networkdrive.log
$ zip networkdrive.zip networkdrive.log
```

tim-felix-lange, 2017-02-14:
@xipmix The problem still consists in 2.041. 
As requested I set up a clean volumio, set up the network drive (took me a few tries to get the path right) and dumped the log as [networkdrive.zip](https://github.com/volumio/Volumio2/files/773380/networkdrive.zip).

As I can't see the false mounts, how do I delete them? Is `rm folder` and remove entries from `data/configuration/networkfs/config.json` sufficient? 

Feature request: Do not log passwords in plain text...

xipmix, 2017-02-14:
Here's the problem
```
Feb 13 20:51:06 volumio volumio[934]: info: Adding a new share
Feb 13 20:51:06 volumio volumio[934]: info: No correspondence found in configuration for share Sinas QNAP on IP 192.168.0.17
Feb 13 20:51:06 volumio sudo[2336]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/mount -t nfs -o ro,soft,noauto 192.168.0.17:/share/M
ultimedia/Sina/Musik/ /mnt/NAS/Sinas QNAP
Feb 13 20:51:06 volumio sudo[2336]: pam_unix(sudo:session): session opened for user root by (uid=0)
Feb 13 20:51:07 volumio sudo[2336]: pam_unix(sudo:session): session closed for user root
Feb 13 20:51:07 volumio volumio[934]: exec error Error: Command failed: /usr/bin/sudo /bin/mount -t nfs -o ro,soft,noauto 192.168.0.17:/share/Multimedia/Sina/Musik/ /mnt/NAS/Sinas\ QNAP
Feb 13 20:51:07 volumio volumio[934]: mount.nfs: mount point /mnt/NAS/Sinas QNAP does not exist
```
As a workaround, if you use an underscore in the name (Sinas_QNAP), it should work.

@volumio: At first I didn't understand why it's failing. The failure happens in the function ControllerNetworkfs.prototype.mountShare in  app/plugins/system_controller/networkfs/index.js.
The space in the mountpoint path seems to be properly escaped:
```
var mountidraw = config.get('NasMounts.' + shareid + '.name')
var mountid = mountidraw.replace(/ /g,"\\ ");
...
var mountpoint = '/mnt/NAS/' +  mountid;
mountutil.mount(pointer, mountpoint, {"createDir": true, "fstype": fstype, "fsopts": fsopts}, function (result) {
### kaboom ###
```
I was loath to believe this was a bug in linux-mountutils, so I wrote a test case:
```
var mountutil = require('linux-mountutils');

var pointer = '192.168.0.2:/var/tmp';
// works
var mountid = "/mnt/a\\ b";
// doesn't
//var mountid = "/mnt/a\ b";
var fstype = 'nfs';
var fsopts = 'soft,ro,noauto'
mountutil.mount(pointer, mountid, {"createDir": true, "fstype": fstype, "fsopts": fsopts}, function (result) {
        if (result.error) {
            console.log(result.error)
        } else {
            console.log("success!");
        }
    });
```
So I think the problem is the \ before the space needs additional escaping to survive the string processing in linux-mountutils.
```
  var mountidraw = "/mnt/a b";
...
- var mountid = mountidraw.replace(/ /g,"\\ ");
+ var mountid = mountidraw.replace(/ /g,"\\\\\\ "); /* yes, Ranger, _6_ backslashes */
```
The reason for 6:
 * each \ we want to insert with the regex needs its own \ in front
 * we get a string that looks like: '/mnt/a\\\ b', ie a space with an escaping \ and a \ with its own escaping \.

xipmix, 2017-02-14:
p.s. great job on the logs Tim, thanks for your efforts.

xipmix, 2017-02-14:
Hm. A bit more testing shows I am completely wrong about the extra escapes, if I test with ```"/mnt/a  b c"``` (ie more than one consecutive space).

I experimented with hacking mountutils.js so it quotes the path before mount/umount
```
--- node_modules/linux-mountutils/mountutils.js.orig    2017-02-14 23:44:03.245297602 +1100
+++ node_modules/linux-mountutils/mountutils.js 2017-02-14 23:44:36.521087982 +1100
@@ -101,7 +101,7 @@
       (options.readonly?"-r ":"") + 
       (options.fstype?"-t " + options.fstype + " ":"") +
       (options.fsopts?"-o " + options.fsopts + " ":"") +
-      dev + " " + path;
+      dev + " " + '"' + path + '"';
 
   // Let's do it!
   var mountProc = exec(cmd, function(error, stdout, stderr) {
@@ -140,7 +140,7 @@
   // Build the command line
   var cmd = (options.noSudo?"":
       (options.sudoPath?options.sudoPath:"/usr/bin/sudo")+" ") + 
-      (options.umountPath?options.umountPath:"/bin/umount") + " " + path;
+      (options.umountPath?options.umountPath:"/bin/umount") + " " + '"' + path + '"';
 
   // Let's do it!
   var umountProc = exec(cmd, function(error, stdout, stderr) {

```
Now the test case works without any escaping at all:
```
var mountutil = require('linux-mountutils');

var pointer = '192.168.0.2:/var/tmp';
// works now
var mountid = "/mnt/a  b c  d";

var fstype = 'nfs';
var fsopts = 'soft,ro,noauto'
mountutil.mount(pointer, mountid, {"createDir": true, "fstype": fstype, "fsopts": fsopts}, function (result) {
        if (result.error) {
            console.log(result.error)
        } else {
            console.log("success!");
        }
    });

```
But the 'fix' is incomplete. The ```isMounted``` function does this check
```
  for (i in mtab) {                                                             
    var mountDetail = mtab[i].split(" ");                                       
    // Does the appropriate field match?  Exact match only.                     
    if ((isDevice && (mountDetail[0]==path)) || (!isDevice && (mountDetail[1]==path))) {
```
which is bogus because the mtab entry is:
```
192.168.0.2:/var/tmp /mnt/a\040\040b\040c\040d nfs ro,soft,vers=4,addr=192.168.0.2,clientaddr=192.168.0.9 0 0
```
Presumably the spaces in ```path``` should be converted to ... octal?  in the same way before comparing with ```mountDetail[1]```.

Possibly the best thing to do here is to replace spaces with underscores at the point where the user inputs the mount path, in ControllerNetworkfs.prototype.addShare.

macmpi, 2017-02-14:
Thanks @xipmix for all these fixes on multiple NAS paths handling issues.
Maybe it's good opportunity to integrate your 5 other pending PRs from November (#898, #883, #879, #877, #876) on somehow related issues.

xipmix, 2017-02-15:
On Tue, Feb 14, 2017 at 05:22:06AM -0800, macmpi wrote:
> Thanks @xipmix for all these fixes on multiple NAS paths handling issues.
> Maybe it's good opportunity to integrate your 5 other pending PRs from November (#898, #883, #879, #877, #876) on somehow related issues.

I'd be happy for that to happen. How do we proceed?

xipmix, 2017-02-15:
A bit more rtfm reveals that the string in /etc/mtab is escaped by {get,set}mntent(7)

> ... octal escapes are used to represent the four characters
> space (```\040```), tab (```\011```), newline (```\012```) and backslash (```\134```).

I thought briefly of trying to do this in ControllerNetworkfs.prototype.addShare but trying to get all the encoding/decoding just right is going to generate lots of bugs, particularly with javascript's ... idiosyncratic support for octal. I think we need to avoid the use of such characters in the mount point entirely.

```
- var mountid = mountidraw.replace(/ /g,"\\ ");
+ var mountid = mountidraw.replace(/[\s\n\\]/g,"_");
```
mountids constructed like this will make it into /etc/mtab unscathed, so the mountutil.isMounted() check should work properly. The user won't care, they never see the mount path, the Nasmounts.shareid.name field is what is shown in the UI.

For the path on the remote device, we can do the space-escaping as is done now, because mountils does not compare ```path``` (escaped or not) with ```mountDetail[0]``` in this case.

PR #1010 (2017-02-16): Avoid unsuitable characters in mountpoints
Issue #684 revealed that linux-mountutils, which we use for mounting NAS drives,
does not handle mountpoints with spaces in them very well - it fails to take
into account the octal encoding of some characters in /etc/mtab (details in
getmntent(7)). This can lead to failures to create mountpoints and failures
to detect that a given share is mounted.
Change how the mountpoint path is constructed from the share name - replace any
characters that get encoded with an underscore. Use this method consistently
in all functions that have to mount or unmount a  [... cut]
closed 2017-02-16 by volumio