volumio/Volumio2 #701: Hifiberry DAC+ and mpdroid: control volume

kiwiASRALL, 2016-09-15:
Hi all,

I just bought the Hifiberry DAC+. It works very well with volumio !

With my phone, I use mpdroid to control volumio at home. Before the DAC, I can control volume with application without any problem. 

Now the control volume has no effect. I should specify the control on the **WebUI works correctly**. The issue is only with a mpd client.

My configuration is here:
[image]
[image]
```

volumio@volumio:~$ cat /boot/config.txt 
initramfs volumio.initrd
gpu_mem=16
force_turbo=1
disable_splash=1
dtparam=audio=on
dtparam=i2c_arm=on
dtoverlay=hifiberry-dacplus
```

Maybe, my configuration is wrong ?

Regards,

volumio, 2016-09-15:
No, it's just that Volumio controls the volume directly, and not via MPD. Its a known issue, and to solve it we must complete the mpd emulation plugin...

kiwiASRALL, 2016-09-15:
Ok thank you :)

HolyMacarony, 2016-09-30:
I would appreciate that too.  Same problem here

closed 2017-06-02 by volumio